AeroPython/PyFME #3: Comunicación con Flight Gear

AlexS12, 2015-09-23:
El objetivo inicial del simulador es menos ambicioso que desarrollar un simulador con gráficos y, en principio, se centra en el motor de mecánica de vuelo. Sin embargo, ya que [Flight Gear](http://www.flightgear.org/) es un simulador libre que, aunque por defecto, trabaja con [JSBSim](http://jsbsim.sourceforge.net/) como motor de mecánica de vuelo, permite elegir otro, sería muy interesante buscar la forma de integrar nuestro trabajo con este motor gráfico. 

Antecedentes: http://es.mathworks.com/help/aeroblks/view-simulation-results.html

AlexS12, 2015-12-25:
Un buen sitio donde buscar información sobre este tema es la wiki de FlightGear:

* http://wiki.flightgear.org/Main_Page
* http://wiki.flightgear.org/FDM_engine_feature_standardization

Juanlu001, 2015-12-26:
Para quien esté más a gusto navegando en GitHub (¿yo?) hay un mirror un poco desactualizado: https://github.com/FlightGear/flightgear

Juanlu001, 2015-12-26:
La clave para la comunicación con un FDM (_flight dynamics model_) externo es esta:

http://wiki.flightgear.org/Property_Tree/Sockets#Using_Proprietary.2FExternal_FDMs

El **property tree** es el árbol de propiedades de FlightGear, que puede ser leído y manipulado usando varios protocolos de red (Telnet, UDP, HTTP). El proceso consiste en llamar a FG con la opción `--fdm=null` para deshabilitar el FDM interno, y especificar el protocolo de comunicación con el FDM externo. Esta es la forma en la que se llama al FDM de MATLAB, por ejemplo.

Aquí están listados todos los métodos para manipular el _property tree_ http://wiki.flightgear.org/Property_tree#Customizing_and_manipulating_the_property_tree_through_Nasal con enlaces a las partes relevantes de la documentación.

Puntos de partida interesantes:

* Echar un vistazo al servidor web que viene incluido con FG http://wiki.flightgear.org/Property_Tree/Web_Server
* Jugar con el lenguaje de scripting de FG, Nasal http://wiki.flightgear.org/Nasal_scripting_language
* Lanzar FG sin FDM y mandar comandos manualmente para ver la respuesta del sistema en tiempo real

elrond30, 2016-08-18:
Tras revisar la documentación de FG, primero es necesario definir que parámetros (modelo de avión, viento, velocidad, etc) se le van a pasar a FG desde PyFME. Una vez están claros, ya se puede mirar la forma y el formato de pasárselos.

elrond30, 2016-08-22:
Leyendo más información, el avión va a ser necesario representarlo en xml. En el siguiente enlace, se puede ver como se hace la llamada y la linea necesaria en el xml:

https://sourceforge.net/p/jsbsim/mailman/message/24380031/

elrond30, 2016-08-22:
Más información al respecto, un buen ejemplo sencillo, de como interactua con FG:

http://diydrones.com/profiles/blogs/closed-loop-autopilot-1

Hay que configurar ficheros xml, para especificar que parámetros se le va a pasar y el formato. Al final del artículo vienen los ficheros.

AlexS12, 2016-08-22:
Interesante, la verdad. Me alegra ver que podemos ir arrojando algo de luz a este tema :smile: 

Yo propondría migrar alguno de los ejemplos que hay a Python o hacer el más básico que se nos pueda ocurrrir. Una vez tengamos eso funcionando creo que el camino será mucho más fácil, no?

http://diydrones.com/profiles/blogs/closed-loop-autopilot-1

podíamos probar con este que ha mandado @elrond30