AeroPython/PyFME #10: Logo

AlexS12, 2015-12-21:
¿Alguien se anima a hacer un logo?

JuanMatSa, 2015-12-21:
Veo que se me ha asignado... Subiré las ideas que tuve, pero cualquier
sugerencia es bienvenida :)

Juanlu001, 2015-12-21:
Si me visitan las musas subo algo :)

JuanMatSa, 2015-12-22:
bueno, algunos diseños que he ido haciendo... 
están a mano sí pero a lo mejor de ahí sacamos los: no me gusta ninguno, me gusta algo este, o espera que tengo una idea a partir de uno de ellos. mojaros, y despachaos a gusto (no soy diseñador gráfico, que me digáis que no os mola no me va a deprimir =D)

un saludo y feliz comiendo de vacaciones
[image]
[image]
[image]
[image]

JuanMatSa, 2015-12-22:
[image]
[image]
[image]
[image]
[image]

Juanlu001, 2015-12-23:
Me gusta mucho la idea de los aviones y la "estela", y el antepenúltimo, el del avión de papel, el que más. Esta semana intento darle una pasadita en Inkscape.

olrosales, 2015-12-23:
A mi el del avión de papel me gusta, pero también  me gusta la idea de los instrumentos de navegación

olrosales, 2015-12-23:
Tengo una foto de una bastón bola pero no puedo subirla desde el movil

olrosales, 2015-12-23:
Se que está muy mal hecho, pero yo propongo algo así,
[image]

Juanlu001, 2015-12-24:
Se ve un poco mal Óscar, es un "bastón y bola" ¿no? Enlaza por aquí si tienes alguna fotillo que sirva de inspiración.

AlexS12, 2015-12-26:
A mí me gusta bastante el avión de papel con la estela y también me molaría algo de este estilo:

[image] con un avioncito superpuesto o algo así.

olrosales, 2015-12-26:
Me gusta mucho la idea de Añex

AlexS12, 2016-01-07:
Estuve jugando un poco el otro día intentando hacer un logo.

[image]

Os dejo el link al svg: https://www.dropbox.com/s/zd092b7b0u0mflg/drawing.svg?dl=0
Por cierto, el avión no es mío... Es una imagen vectorial libre.

Juanlu001, 2016-04-14:
La versión 0.1 tendrá que tener su logo :)

Juanlu001, 2016-07-07:
Puedo prometer y prometo que PyFME v0.1 tendrá logo.

Juanlu001, 2016-11-06:
¿Qué os parece? Basado en la idea de @JuanMatSa :grin:

[image]

commit c2c5eda0 (2016-11-06): Add logo, close #10
closed 2016-11-06 by Juanlu001