AeroPython/PyFME #4: json vs xml

AlexS12, 2015-12-21:
JSBSim utiliza ficheros xml para crear sesiones, aeronaves, estado inicial... Evaluar ventajas e inconvenientes con respecto a json.

Juanlu001, 2015-12-21:
"Off the top of my head":

En general los ficheros XML suelen ser mucho más verborreicos y densos que los JSON, que son más ligeros. Por contra, los ficheros XML tienen la ventaja de que se pueden validar contra un esquema preestablecido, para asegurar que tienen la estructura deseada.

No tengo datos sobre cuál es más rápido de parsear, pero no creo que sea un criterio de diseño importante, puesto que por lo que dices se van a leer muy pocas veces.

Por otro lado, si bien XML es más difícil de escribir por humanos sin la ayuda de editores, JSON tampoco es extremadamente amigable. YAML es un formato pensado para ser escrito por humanos, y se parece bastante a JSON, si bien no tiene soporte en la biblioteca estándar.

Por último, si la intención es leer ficheros de configuración ya existentes de otros programas y XML es su formato preferido, la elección es clara: amoldarse a XML.

Software relacionado:

* XML: LXML o biblioteca estándar.
* JSON: simplejson o biblioteca estándar.
* YAML: PyYAML

Juanlu001, 2015-12-21:
Con lo de validación me refiero a esta línea: http://sourceforge.net/p/jsbsim/code/ci/f1c390af/tree/aircraft/A320/A320.xml#l5 Es la que asegura que el XML está «hablando de aviones» y que no va a haber problemas al leerlo.

newlawrence, 2015-12-27:
Como comenta @Juanlu001, los ficheros **XML** pueden validarse para asegurar que tienen la estructura deseada; si nos decantásemos por **JSON** tendríamos que escribir bastante más código de validación pues **JSON** no es más que un formato rápido para almacenar estructuras de datos medianamente complejas en **JavaScript**.

Aún así, eso no quita que en las partes en las que la biblioteca se comunique con otras (escribo con **FlightGear** en mente), dichos ficheros de intercambio sería más conveniente que se generaran en formato **JSON** ya que en este tipo de situaciones es casi un estándar.

Resumiendo, mi propuesta es:
* **XML** para ficheros de configuración, definición y datos de los problemas a resolver.
* **JSON** para ficheros de intercambio (en caso de necesitarlos).

AlexS12, 2015-12-27:
A ver! A esto le he estado dando una vuelta yo también y m  conclusión principal es *En FlightGear tirán de este tipo de ficheros porque C++ es un lenguaje compilado*. Tu quieres correr nuevos casos e incluir aviones sin compilar o incluso sin conocer el código. Sin emabrgo:

* Para nosotros es más fácil, más divertido y creo que hasta más bonito crear nuestros casos con scripts de Python (gracias a las clases que no tenemos, podemos tener una clase `Case` o `Simulation` o como se llame en la que se configuren los atributos que hagan falta para un caso de simulación. De este modo las simulaciones se configuran en Python y no acabamos haciendo un programa python que genere un xml que lea un programa python para correr el simulador en python...

* Con los aviones pasa un poco parecido. Si los miráis detenidamente, los xml que definen los aviones lo que hacen es "definir funciones"... No sólo valores de parámetros, ya que cada avión lleva una aerodinámica distinta. No se trata de tener siempre las mismas funciones y alimentarlas con unos coeficientes, sino que dependiendo del avión las funciones y el número de coeficientes variarán. Para nosotros es mucho más sencillo que esas funciones vayan directamente sobre un `.py` que sea el que contenga el modelo de avión.

No sé si me explico o si a lo mejor a pesar de tener un lenguaje de scripting es mejor apartar ciertas cosas del código. Pero vamos yo creo que podemos deshacernos del xml y probablemente del json.

newlawrence, 2015-12-27:
¡Muy buena idea @AlexS12 😎!.

A lo que te refieres es a darle una estructura de plugins a los supuestos. No es complicado. Si se definen unas clase base que describan el problema más sencillo posible (esto es, un problema vacío y una instrucción para correrlo), para describir un supuesto bastaría con heredar dichas clases y modificar/añadir los métodos y propiedades que se deseen.

El motor de la biblioteca lo único que tiene que hacer entonces es tener una lógica para buscar y ejecutar dinámicamente esos ficheros (mecanismo similar al que utilizamos el año pasado para hacer los [benchmarks de los modelos **ISA**](https://github.com/AeroPython/aeropy/blob/alberto-cpp/test/performance.py)).

AlexS12, 2015-12-28:
Pues voy a cerrar este issue y vamos discutiendo estos temas en arquitectura! Gracias por vuestras aportaciones. :grinning:

closed 2015-12-28 by AlexS12