AeroPython/PyFME #5: Creación del avión de prueba

AlexS12, 2015-12-21:
Puede ser definido en un archivo .py con sus propiedades másicas, aerodinámicas... escritas en él.
__Hay que definir como se cargan los datos de este avión en el archivo main de prueba__. 

Se pueden obtener datos de alguno de los aviones ya creados en FlightGear almacenados en xml's: http://sourceforge.net/p/jsbsim/code/ci/master/tree/aircraft/ pero debe ser un avión sencillo.

 
Más adelante se pasará a un archivo con el formato acordado y se creará una clase que lo cargue.

olrosales, 2015-12-24:
A mi no me importa pegarme un poco con este archivo y ver los coeficientes que disponemos y hacer el avión de prueba. Lo único, ¿que avión utilizo?

AlexS12, 2015-12-24:
Parece que echando un vistazo en estos sitios:

* http://m-selig.ae.illinois.edu/apasim/Aircraft-uiuc.html
* http://longfly.square7.ch/

La Cessna 172p Skyhawk:

* http://wiki.flightgear.org/Cessna_172P
* https://en.wikipedia.org/wiki/Cessna_172

Parece una buena opción ya que se disponen de bastantes datos. No obstante si te animas a echar un vistazo y te parece mejor algún otro modelo, siéntete libre.

He estado mirando y por lo que parece en JSBSim es el propio avión el que lleva sus funcioens para devolver las fuerzas y momentos. Así que, yo creo que no es descabellado que lo hagamos igual. Podemos fusionar este issue con el #8.

Por cierto, ¡te voy ayudando con esto también!

olrosales, 2015-12-25:
Vale gracias Alex, de aquí ( http://sourceforge.net/p/jsbsim/code/ci/master/tree/aircraft/c172p/c172p.xml ) me cuesta muchísimo interpretar los datos. Extrayendo los datos de ( http://m-selig.ae.illinois.edu/apasim/Aircraft-uiuc/cessna172-v1/ ) creo que están todos pero no sé bien interpretar el CD.

Juanlu001, 2016-01-04:
No entiendo por qué sale que quité el issue a Óscar, no lo hice conscientemente.

AlexS12, 2016-01-04:
Ha salido en varios, que has quitado a la persona asiganada... jajajajaja No sé por qué será.

Pasamos a este issue las conversaciones sobre el avión que estamos definiendo, su modelo aerodinámico, de motor... 

@olrosales haz el pull request aunque no esté lista del todo la rama y vamos revisando.

olrosales, 2016-01-08:
Tengo una duda con respecto a los test, si las funciones aerodinámicas vienen dentro de los archivos del avión "aircraft.py",¿Habría que hacer un test para cada avión? Tengo pensado una forma de hacer uno genérico pero no se llamar a datos que estén dentro un archivo diferente. ¿Es tan sólo hacer import aircraft? Y aún así habría que cambiar esa línea para testear cada avión.

Juanlu001, 2016-01-08:
Yo creo que cada avión tendría que tener sus tests, pero a ver qué opina @AlexS12

AlexS12, 2016-01-08:
@Juanlu001 De momento sí. Es cierto que luego podemos sacar a un módulo de aerodinámica determinadas funciones que usen todos o mucho de los aviones para calcular las fuerzas. Pero vamos a esperar un poquito hasta que seamos capaces de identificar bien esas partes generalizables del código. Si tú ya tienes localizadas algunas apúntalas y para la próxima reunión lo comentamos.

AlexS12, 2016-01-10:
cierro el issue y seguimos discutiendo en pull request #23 sobre el código. :airplane:

closed 2016-01-10 by AlexS12