AeroPython/PyFME #6: Escribir ecuaciones a integrar

AlexS12, 2015-12-21:
Escribir en funciones las ecuaciones que deben ser integradas:
* Fuerzas y momentos.
* Relaciones cinemáticas angulares.
* Ecuaciones de navegación.

Cada grupo de ecuaciones anteriores deber estar en una función distinta, ya que se pueden integrar secuencialmente.

Definir inputs, outputs, __nomenclatura__.

@DLpadilla había empezado esta parte.

AlexS12, 2016-01-01:
Podéis seguir el progreso de esto aquí: https://github.com/AlexS12/PyFME/tree/dynamic-equations Está en un fork mío porque Dani anda todavía apañandose con GitHub pero casi todo se lo ha escrito él.

Faltan por documentar las funciones y escribir algunos tests. Vamos a trabajar en ello estos días. Echandle un ojo.

¿Qué os parecería cambiar los `np.cos` y `np.sin` por sin y cos haciendo al principio del script `cos = np.cos` y `sin = np.sin`. Sé que no es el estándar, pero yo lo usé para las coordenadas porque limpia mucho las ecuaciones y evita incluso mucho saltos de línea por líneas largas.

¿Más sugerencias?

Juanlu001, 2016-01-01:
@AlexS12 A falta de hacer una revisión más en detalle, yo haría 'from numpy import sin, cos' que al final es lo mismo

AlexS12, 2016-01-10:
Cierro el issue y seguimos discutiendo sobre el pull request #18. Las ecuaciones se pueden considerar completas (a falta de algún test) @DLpadilla está en ello.

closed 2016-01-10 by AlexS12