AeroPython/PyFME #7: Creación del main.py

AlexS12, 2015-12-21:
Este fichero __provisional__ será el que utilicemos para hacer las primeras simulaciones.

En él se cargarán los datos del avión, se configurará el estado inicial y se irán integrando las ecuaciones. 
* Las fuerzas (aerodinámicas, gravitatorias, motor) serán generadas por funciones externas. 
* Las ecuaciones estarán definidas en funciones externas.
* Las funciones auxiliares (como cambios de sistema de referencia estarán definidas por funciones externas.
* Las funciones relacionadas con atmósfera estarán también definidas externamente.

Se pueden definir varias maniobras de prueba.

AlexS12, 2016-01-04:
Este main.py se ha convertido en `src/pyfme/example_001.py` que a lo mejor deberíamos meter en una carpeta, al nivel de src, `PyFME/examples/example_001.py`. De momento está en el pull request #18.

Juanlu001, 2016-01-04:
:+1: Justamente lo iba a decir cuando pidiérais una revisión del PR.

AlexS12, 2016-01-10:
Cierro el issue y seguimos discutiendo sobre el pull request #18.

closed 2016-01-10 by AlexS12