AeroPython/PyFME #8: Modelo de fuerzas aerodinámicas

AlexS12, 2015-12-21:
Ecuaciones que proporcionen las fuerzas aerodinámicas (dimensionales o adimensionales), dadas las deflexiones de mando (y presión dinámica y superficie mojada en caso de que sean dimensionales).

Modelo lineal, estacionario, para avión rígido.

olrosales, 2015-12-23:
A mi esta tarea me podría interesar, igual si alguien se anima conmigo porque tengo algunas dudillas que me gustaría poder comentar con alguien.

AlexS12, 2015-12-23:
Comenta por aquí tus dudas. Había pensado definir en la reunión las cabeceras de este tipo de funciones, pero podemos empezar a hacerlo aquí (así queda constancia y el que lo necesite se entera).

Ve pensando en una función que podamos llamar de este modo (tengo otras ideas en mente pero los cambios sería sencillos):

    calculate_aerodynamic_forces(alfa, beta, delta_e, delta_a, delta_r, aircraft)
    calculate_aerodynamic_moments(alfa, beta, delta_e, delta_a, delta_r, aircraft)

Esperaría que devolviese las tres componentes de la fuerza en ejes viento. Podría calcular la cantidad adimensional (ya nos encaragaríamos luego de darle dimensiones) y podemos, en principio, si resulta demasiado complicado, dejar las componentes debidas a las velocidades angulares y derivadas temporales de alfa, beta, deltas... para más adelante.

Asume que los coeficientes van a venir "dentro" de aircraft y que los vas a poder llamar de este modo:
    
    c_L_0 = aircraft.c_L_0
    c_L_alpha = aircraft.c_L_alpha

Como veis, esto implica ponerse de acuerdo con el que haga el avión de prueba #5 para que tengas disponibles las variables y las llaméis igual. Te puedes encargar tú también de hacerlo y así tienes los coeficientes disponibles.

--

Tengo pendiente, pensar y lanzaros la pregunta de dónde debe ir el modelo de fuerzas aerodinámicas. Cuando lo orientemos a objetos, igual debería ser el propio avión el que llevase sus métodos de calcular las fuerzas, ya que éstas tendrán que adaptarse a los datos aerodinámicos de los que dispongamos en cada caso (que serán más o menos según el avión). Pongo por aquí a @Juanlu001 y @newlawrence que creo que pueden tener alguna idea al respecto, para que arrojen luz cuando llegue el momento.

olrosales, 2015-12-24:
@AlexS12 A continuación adjunto (en una foto) lo que podría ser la fuerza de sustentación adimensional, rígido, estacionario, estático y lineal, que sería parte de la salida de las fuerzas, por ahora he decidido considerar las simplificaciones propuestas en el libro de mecánica del vuelo (pag 205 - 207), no obstante si no os parece bien es tan sólo añadirle términos.  Además cabe destacar que **el origen de ángulo de ataque está situado en la LSN**, es importante saber si los coeficientes que utilicemos están igual.

El coeficiente de momentos también tiene algunas simplificaciones y si no os parece considerarla las podemos escribir desde el principio.

Si todo esto os parece bien, termino de sacar el modelo y me pongo a escribirlo

[image]
[image]

AlexS12, 2015-12-24:
Ten en cuenta que esos coeficientes son los que vamos a coger de los sitios, con lo cual se puede ver en el modelo cómo están, pero que no vas a necesitar las fórmulas con las incidencias y efectividades. Sólo en caso de que no dispusiésemos del coeficiente, tendríamos que modelarlo con esas fórmulas, en el resto de situaciones serán valores únicos o tablas de valores sobre las que interpolemos.

AlexS12, 2016-01-04:
Pasamos esta conversación al issue #5

closed 2016-01-04 by AlexS12