AeroPython/PyFME #12: __init__.py en carpeta de test

AlexS12, 2015-12-24:
Según las [indicaciones de `pytest`](https://pytest.org/latest/goodpractises.html), no es recomendable tener un __init__.py en la carpeta de test. 

> avoid “__init__.py” files in your test directories. This way your tests can run easily against an installed version of mypkg, independently from the installed package if it contains the tests or not.

No obstante, no he conseguido correr los test sin él. Al intentarlo daba un error de importación: `no module named pyfme`.

He visto que @Juanlu001 no tiene el __init__.py en la carpeta de tests y que tiene un testing.py en la carpeta principal de `poliastro` y una función `test` en el __init__.py de polisastro. No he conseguido los porqués. 

Ya que estamos con los tests. No sé si la manera más organizada es irlos metiendo en carpetas separadas dentro de cada uno de los subpaquetes o tenéis alguna idea mejor.

Juanlu001, 2015-12-24:
Efectivamente, eso me lo arregló @newlawrence en https://github.com/poliastro/poliastro/pull/67. Veo que ya cambiaste la estructura en 19a4b0cf75aedc821aaf161875386f4ef9e03a69, así que solo falta invocar pytest correctamente. Esta me la asigno yo.

En cuanto a la estructura de los tests, ya aviso de que con esto no hay consenso. Las dos visiones son: incluir los tests junto al código o no. Cada una de las dos tiene sus desventajas e inconvenientes. A mí en poliastro me ha venido muy bien incluirlos porque gente no muy hábil con GitHub me ha mandado informes de fallos gracias a eso. Tal y como está ahora está perfecto, no hace falta tocarlo más.

Juanlu001, 2015-12-26:
Ah, veo que no contesté a la pregunta:

Tener ese módulo `testing` en poliastro e importarlo en `__init__.py` es para poder hacer `python -c "import poliastro; poliastro.test()"` o bien

```
>>> import poliastro
>>> poliastro.test()
```

Es parecido a lo que trae NumPy y SciPy. La otra opción sería no incluir los tests, y cuando alguien quisiera correrlos tendría que instalar el paquete, luego bajarse las fuentes del repositorio y llamar a `py.test`. Es un paso más, pero puede ser uno importante.

El hecho de que las carpetas de tests no tengan `__init__.py` es para que Python no los detecte como paquetes, porque en realidad no es código que vayas a ejecutar. El efecto de eso es que los tests y el código en sí son independientes, que es lo que dice el manual de buenas prácticas.

Por cierto, para correr los tests de PyFME no he tocado nada, simplemente hay que invocar `py.test` sin argumentos:

```
(pyfme35)[juanlu@nebulae]─[~/Development/Python/AeroPython/PyFME]
└[11:09:55] $ py.test
============================= test session starts ==============================
platform linux -- Python 3.5.1, pytest-2.8.1, py-1.4.30, pluggy-0.3.1
rootdir: /home/juanlu/Development/Python/AeroPython/PyFME, inifile: 
collected 3 items 

src/pyfme/utils/tests/test_coordinates.py ...

=========================== 3 passed in 1.20 seconds ===========================
```

Si te das por satisfecho con esto puedes cerrar la incidencia :)

AlexS12, 2015-12-26:
Claro! corren porque tienen el `__init__.py` en la carpeta de tests. Si se lo quito me dice que no encuentra pyfme.

```
asaez@asaez-Satellite-A660 ~/aero/PyFME $ py.test
============================= test session starts ==============================
platform linux -- Python 3.4.3, pytest-2.8.1, py-1.4.30, pluggy-0.3.1
rootdir: /home/asaez/aero/PyFME, inifile: 
collected 0 items / 1 errors 

==================================== ERRORS ====================================
__________ ERROR collecting src/pyfme/utils/tests/test_coordinates.py __________
src/pyfme/utils/tests/test_coordinates.py:12: in <module>
    from pyfme.utils.coordinates import (body2hor, hor2body,
E   ImportError: No module named 'pyfme'
=========================== 1 error in 0.36 seconds ============================
```

Supongo que de alguna manera la estoy liando con los módulos.

Juanlu001, 2015-12-26:
Cierto, no me había dado cuenta (se nota que todavía me cuesta entender esto). Este error es lo que uno busca cuando mete todo el código en la carpeta `src`: que no se solape con el código instalado.

Lo que hay que hacer en este caso es instalar PyFME en modo editable. El problema es que ahora mismo no hay un `setup.py` disponible, y por eso no se puede instalar:

```
(pyfme35)[juanlu@nebulae]─[~/Development/Python/AeroPython/PyFME]
└[13:35:47] $ pip install -e .
Directory '.' is not installable. File 'setup.py' not found.
```

Voy a abrir un pull request con el mínimo código que hace falta para que esto funcione.

AlexS12, 2015-12-26:
Ok! gracias Juanlu!

commit d210df88 (2015-12-26): Add minimal setup.py, fix #12
* Project name
* Project version (development), PEP 440 compliant
* Packages
* Package dir, see
https://docs.python.org/3/distutils/setupscript.html#listing-whole-packages
closed 2015-12-26 by Juanlu001