AeroPython/PyFME #13: Decidir una licencia

Juanlu001, 2015-12-26:
Ahora mismo ya hay código escrito pero no tiene licencia.

http://choosealicense.com/licenses/

> You retain all rights and do not permit distribution, reproduction, or derivative works. You may grant some rights in cases where you publish your source code to a site that requires accepting terms of service. For example, publishing code in a public repository on GitHub requires that you allow others to view and fork your code.

Sería conveniente elegir alguna. Por ejemplo, JSBSim es Lesser GPL versión 2:

http://sourceforge.net/p/jsbsim/code/ci/master/tree/COPYING

Por otra parte, lo más habitual en proyectos de Python es usar MIT o BSD de 2 cláusulas. Aquí por qué cambié de una a otra en poliastro:

https://github.com/poliastro/poliastro/pull/41

Como ya sabéis, se trata de decidir entre un modelo permisivo o uno _copyleft_:

http://es.stackoverflow.com/a/363/74

Juanlu001, 2015-12-26:
Y un aviso para navegantes, concretamente por si alguno está pensando en leer el código fuente de JSBSim: la LGPL tiene una definición de *"work based on the Library"* que incluye traducciones de código. De hecho, según algunas interpretaciones, inspirarse en código LGPL para escribir tu propia biblioteca te forzaría a liberarla bajo la LGPL. Hacer lo contrario sería una violación de la licencia y un problema grave. Qué es "inspiración" y qué es "plagio" es algo deliberadamente ambiguo que solo podría resolver un tribunal.

Soluciones:

* No **leer** código (L)GPL, nunca.
* Liberar PyFME bajo la LGPL y no pensar más en ello.

AlexS12, 2015-12-26:
No me considero experto en el tema, de hecho evito leer acerca de esto porque caigo rendido a la segunda línea. Pero mi opinión es:

* Es estúpido poner grandes restricciones (incluso pequeñas) a un código que está aquí, a la vista de todo el mundo. Si quisiéramos que la gente no lo usase tal cual, modificado o incluido entre su código, haríamos un repositorio privado; creo que nuestro espíritu es el contrario: __Hacer algo con lo que aprendamos nosotros y pueda aprender el resto del mundo y si encima es útil para alguien, mejor, porque más aprenderemos y más nos realizará esto__.

* Aunque aumentásemos las restricciones, nunca tendríamos la capacidad de comprobar si alguien se las está saltando. En este sentido tendremos que correr con el riesgo de que los de SpaceX empiecen a forrarse con nuestros avances.

* Respecto a lo de LGPL, me preocupa más que la inspiración, otra cosa. Qué pasaría si intentásemos enchufar el simulador a FlightGear. Porque supongo que usan la misma licencia...

Juanlu001, 2015-12-26:
Sobre tu tercer punto: precisamente la clave del asunto es la definición de ["enchufar"](http://programmers.stackexchange.com/q/167773/15297). Aquí en ese sentido no hay peligro, porque:

* FG puede enchufar PyFME casi sin problemas, sobre todo si lo escribimos como un programa externo que se comunica a través de la red (ver #3).
* PyFME no va a enchufar FG porque la relación de dependencia es al revés, y por tanto al primero no le afecta la licencia del segundo.

AlexS12, 2016-01-04:
Nadie más se ha pronunciado por aquí... A mí la licencia MIT me parece bien. 

Descartaría la LGPL porque podría haber gente interesada en usar algunas de nuestras funciones auxiliares, la ISA... para otro tipo de bibliotecas y estaríamos imponiéndoles la licencia. Por otro lado, estamos tirando siempre de libros, papers... por lo que no creo que haya ningún problema con JSBSim.

¿Alguien tiene algo que añadir?

olrosales, 2016-01-04:
@AlexS12 Yo no entiendo nada sobre licencias, la que mejor veais

Juanlu001, 2016-01-11:
Me asigno yo subir una licencia esta semana.

commit 7028974f (2016-01-13): Add license, fix #13
closed 2016-01-13 by Juanlu001