AeroPython/PyFME #21: Consistencia en las cabeceras de las funciones - pasar arrays o números

AlexS12, 2016-01-05:
De momento no pasa nada (lo pongo para aquí para acordarnos), pero para la siguiente vez que nos juntemos deberíamos llegar a un acuerdo sobre como pasar los argumentos que son vectores, si como arrays o por cada una de sus componentes. De momento tenemos ejemplos de ambas:

    linear_and_angular_momentum_eqs(time, vel, mass, inertia, forces, moments)

    body2hor(body_coords, theta, phi, psi)

otra opción más avanzada, como comentaba @Juanlu001 en el grupo de Telegram, sería hacer algo parecido a lo que hace SymPy o usarlo directamente.

AlexS12, 2016-01-13:
El consenso es que siempre que sea posible pasaremos los vectores, en lugar de las componentes. ls nombres recomendados son:

lin_vel = (u, v, w)
ang_vel = (p, q, r)
pos = (x, y, z)
attitude = (theta, phi, psi)

alpha y beta se pasarán separados

Juanlu001, 2016-01-14:
¡Esto merece una entrada en la wiki también!

AlexS12, 2016-01-14:
Ok! Me lo apuntó. Reabro el issue para acordarme, igual requiere cambiar cosas en código. Echaré un vistazo también!

Juanlu001, 2016-07-07:
Esto merece una entrada en la wiki @AlexS12 :wink:

AlexS12, 2016-07-09:
Esto afectaría fundamentalmente a las funciones que hicimos al principio del simulados. Ahora, la verdad es que usando poo y pasando las clases que os enseñé por cabecera, se simplifica bastante el tema.

Me apunto la entrada en la wiki! :+1:

Juanlu001, 2016-08-17:
Supongo que esto está relacionado con #41, ¿qué opináis?

AlexS12, 2016-08-17:
Sí, yo creo que vamos a cerrar este y dejamos el otro abierto para no olvidarnos

closed 2016-08-17 by AlexS12