AeroPython/PyFME #24: Creación de modelo de esfera para tests

AlexS12, 2016-01-07:
Vamos a crear además del avión una esfera para ir haciendo tests. Aunque la parte que se refiere a actitud no tenga mucho sentido para una esfera (sobre todo porque como gire locamente vamos a tener [gimbal lock](https://en.wikipedia.org/wiki/Gimbal_lock)) vamos a implementar una para casos de prueba.

La idea es hacer algo parecido al issue #5 (pull request #23). Simplemente habría que definir una función que devuelva la resistencia. Podemos dar sustentación en función de la velocidad angular (hay modelos sencillitos en papers por internet... lo que he econtrado está en nuestras referencias) y si nos venimos arriba podemos dar un momento que provenga de la fricción y que tenderá a disminuir la velocidad angular.

Como siempre, cuanto más mejor, pero pasito a pasito.

AlexS12, 2016-01-08:
@JuanMatSa comenta por aquí lo de los ejes cuerpo, viento... cuando esté claro. Y la refencia de las ecuaciones para que todos puedan echar un ojo porfa.

JuanMatSa, 2016-01-09:
buenos días a todos!

Mientras voy acabando el primer script  y lanzando el pull request aprovecho para empezar a preguntar y comentar algunas cosillas, como pidió @AlexS12:

1º: necesitamos los coeficientes de las fuerzas para algo? (p.e. el coeficiente de resistencia) cuando suba el script lo vereis, pero a priori tengo una funcion que da los coeficientes, las fuerzas y el vector asociado en ejes cuerpo

2º ejes cuerpo??!! Pero si la resistencia aerodinámica ya está en ejes viento! Es cierto, la resistencia aerodinámica (la tenemos tabulada en función del número de Re [1]) y está en ejes viento. Sin embargo la fuerza del efecto Magnus no es tan trivial:
Lo que siempre vimos en clase era el caso 2D:
[image]

En el caso 2D el efecto magnus se corresponde con la sustentación.
Sin embargo en el caso 3D las cosas son un pelín menos claras: Supongamos que los ejes viento coinciden con los cuerpo por simplicidad y sin perdida de generalidad (alpha=0, beta=0). Supongamos además que la pelota gira entorno al eje y: en ese caso, las fuerzas del magnus coincidirían con el eje z y la fuerza del efecto magnus coincide con la sustentación. 
Como los ejes cuerpo y los ejes viento coinciden una rotación entorno al eje x no produce efecto magnus. Sin embargo, si además de rotar en torno al eje y, rota entorno al eje z aparecerá de nuevo el efecto magnus produciendo una fuerza en este caso lateral, que ya no deberíamos llamar sustentación (os acordais del gol de Roberto Carlos?)
[image]
Es decir, la fuerza llevará la dirección del producto vectorial de una velocidad y la otra. y ¿qué modulo?
[image]

En la bibliografía [1] podemos ver (diapositiva 11-12) que la fuerza del efecto magnus viene tabulada (la que aquí llama sustentación ¬¬) y depende de un parametro "effective spin number" Sn: que viene a ser la proyección normal de la velocidad angular sobre la velocidad lineal:
[image]

Es decir, el modulo del producto vectorial entre el modulo de la velocidad lineal.

y de nuevo, por qué en ejes cuerpo??

Bueno, tenía las velocidades en ejes cuerpo del input, así que esta fuerza estaba en ejes cuerpo y había que decidir si pasar a ejes viento el magnus effect o a ejes cuerpo la resistencia. Y dado que en ambas situaciones con tan solo llamar a coordinates se soluciona, decidi en ejes cuerpo (a cambio necesita que le digamos a la función el alpha y el beta)

[1] Aerodynamics of Sports Balls. Bruce D. Kothmann. January 2007

JuanMatSa, 2016-01-09:
Bueno, he actualizado mi fork:
https://github.com/JuanMatSa/PyFME/tree/JuanMatSa_ball/src/pyfme/ball
no hago pull request hasta terminar los tests si os parece bien.

sigo en ello =)

AlexS12, 2016-01-09:
Ok! Sólo una cosa! Mete mejor el archivo ball.py en aircrafts y los tests en una carpeta que se llame tests dentro de aircrafts. A lo mejor en algún momento hace falta una carpeta por avión... pero por ahora yo creo que es mejor así.

JuanMatSa, 2016-01-09:
okey, en el proximo commit :)

AlexS12, 2016-01-10:
cierro el issue y seguimos discutiendo sobre el pull request #28

closed 2016-01-10 by AlexS12