AeroPython/PyFME #25: Definición de dependencias

AlexS12, 2016-01-07:
De momento creo que no es un problema, pero estaría bien para evitar sorpresas, definir el entorno que tenemos que configurar en conda para trabajar todos con los mismos paquetes.

Juanlu001, 2016-01-11:
Lo que podemos hacer es subir un archivo `environment.yml` que reconstruya el entorno conda:

http://conda.pydata.org/docs/using/envs.html#export-the-environment-file

Podríamos acordar trabajar todos en Python 3.5 y dejar que Travis ejecute los tests en otras versiones, por ejemplo.

Juanlu001, 2016-01-11:
O de esta otra forma, que es un poco más flexible y puede servir para reutilizarlo en el entorno de integración continua:

http://conda.pydata.org/docs/using/envs.html#build-identical-conda-environments

AlexS12, 2016-01-13:
Pues si la segunda nos puede venir mejor, podemos decantarnos por esa. ¿Qué opinas?

Juanlu001, 2016-01-16:
Nótese que he hecho referencia a este issue desde un commit. Incluyendo `fixes #NN` en el mensaje, en cuanto se fusione la pull request se cerrará automáticamente esta tarea. Magic! :dizzy:

Juanlu001, 2016-01-16:
Por cierto, para futuras referencias: la creación de entornos con archivos `environment.yml` está un poco mal documentada (haciendo pruebas he visto que son más flexibles de lo que yo pensaba) y de hecho puede que en el futuro eliminen el comando `conda env` (se lo he leído a Mike Sarahan, de Continuum, en algún comentario pero ahora no lo encuentro). De momento ver #42 para más detalles.

AlexS12, 2016-01-16:
Vale, estupendo! Gracias @Juanlu001 buena información!

Woow! lo de fixes me ha matado: https://help.github.com/articles/closing-issues-via-commit-messages/

commit 1c24f214 (2016-01-17): Add Read the docs config and env file
This also fixes #25.
closed 2016-01-17 by Juanlu001