AeroPython/PyFME #26: Tutorial paso a paso del flujo de trabajo

AlexS12, 2016-01-07:
Tarea de documentación que igual nos resulta útil. Id rellenándolo, yo iré añadiendo cosas cuando vaya teniendo algún rato. Igual esto sirve de inspiración: http://suave.stanford.edu/develop/

Aquí la página de la wiki que queremos ir completando:
https://github.com/AeroPython/PyFME/wiki/Tutorial-paso-a-paso-del-flujo-de-trabajo

AlexS12, 2016-01-13:
¡Cerramos cuando esté completo en la wiki!

JuanMatSa, 2016-01-22:
qué sería bueno poner de git? todas las opciones que tiene add commit...? yo creo que no hay mejor explicación que la de gitinmmersion, pero puedo ponerlo un poco resumido con los pantallazos del propio git

AlexS12, 2016-01-22:
@JuanMatSa No te compliques demasiado. git status, git add, git commit, git checkout y poco más..
Usa de ejemplo algún caso que hayas hecho y ya está

AlexS12, 2016-01-25:
Veo que no ha habido cambios en la wiki aquí. Lo dejo abierto!

AlexS12, 2016-02-15:
He añadido un link a http://gitimmersion.com/lab_04.html en la parte sobre add, commit, status... con las lecciones correspondientes. ¡Podemos darlo por cerrado!

closed 2016-02-15 by AlexS12