AeroPython/PyFME #30: Nomenclatura

AlexS12, 2016-01-12:
Definición de la nomenclatura en la [página](https://github.com/AeroPython/PyFME/wiki/Symbols-list) de la wiki (modificar al antojo, todo lo que hay ahora mismo son ejemplos). Llegar a acuerdo por parte de @AeroPython/pyfme 

Véase:

```
CL0, CLa, CLde, CLdih = Long_coef_matrix[0,:]
CD0, CDa, CDde, CDdih = Long_coef_matrix[1,:]
CYb, CYda, CYdr = Lat_coef_matrix[0,:]
```
```
Cmfull = Cm0 + Cma * alpha + Cmde * deltae + Cmdih * ih
Clfull = Clb * beta + Clda* deltaail + Cldr * deltar
Cnfull = Cnb * beta + Cnda* deltaail + Cndr * deltar 
```
```
 m = 4600 * 0.453592   # ¿ponemos unidades aquí?
 Ixxb = 8884 * 1.3558179
 Iyyb = 1939 * 1.3558179
 Izzb = 11001 * 1.3558179
 Ixzb = 0 * 1.3558179
```
```
Sw = 16.258   # ¿Los alargamos?
c = 0.4450056
b = 3.428122
```

Juanlu001, 2016-01-12:
Las unidades sí que me parecería importante ponerlas, en comentarios aunque sea (de momento). Se puede acordar usar siempre el SI salvo en casos en los que los kilómetros sean más útiles que los metros (por ejemplo). Y otra cosa: yo separaría los subíndices del símbolo con un guion bajo - aunque sea más largo de escribir, creo que queda más claro.

```python
C_L0, C_La, C_Lde, C_Ldih = Long_coef_matrix[0,:]
C_D0, C_Da, C_Dde, C_Ddih = Long_coef_matrix[1,:]
C_Yb, C_Yda, C_Ydr = Lat_coef_matrix[0,:]
```

```python
 m = 4600 * 0.453592   # ¿ponemos unidades aquí?
 I_xxb = 8884 * 1.3558179
 I_yyb = 1939 * 1.3558179
 I_zzb = 11001 * 1.3558179
 I_xzb = 0 * 1.3558179
```

AlexS12, 2016-01-13:
Esto ya está en marcha, a partir de ahora discutimos sobre lo de la wiki. Se da por cerrado el issue

closed 2016-01-13 by AlexS12