AeroPython/PyFME #36: Casos de ejemplo y test

AlexS12, 2016-01-15:
Comenzamos con las nuevas tareas!

Por el momento tenemos implementadas las funcionalidades básicas, a falta de que se terminen de fusionar los cambios en #23 (first aircraft @olrosales) y #20 (anemometry @martosc).
Conviene que en cuanto incorporemos estas ramas se pueda empezar a trabajar en algún ejemplo "más serio" que el que tenemos del avión volando. Probablemente nos demos cuenta de algunos fallos...

Podéis proponer maniobras y tratar de implementarlas. Cuantas más tengamos, mejor. Por otro lado estaría bien que todos los que estamos desarrollando hagamos algún caso. De este modo seremos nuestros propios usuarios y podremos ir viendo cómo de cómodo es usar el paquete que estamos haciendo, qué tal funciona, qué necesidades tenemos...

AlexS12, 2016-01-25:
Alguien se ha animado a jugar un poco con esto?

Es importante antes de implementar muchas más cosas probar qué tal funciona lo que tenemos, si es cómodo usarlo, si funciona rápido... 

Ahora que ya está casi todo en el máster es fácil de probar. En cualquier caso, si alguien quería animarse, pero le ha dado reparo por las ramas y demás, le creo una rama de pruebas aquí con todo lo que haga falta fusionado.

AlexS12, 2016-01-26:
Con las ramas que se han mergeado últimamente, **esto ya se puede probar sin necesidad de fusiones raras**. Simplemente nueva rama y a tirar casos de ejemplo.

__El hecho de ponerse con esto nos va a generar bastante tarea nueva__, por ejemplo: 

* las condiciones iniciales que se dan al sistema son: (u, v, w, p, q, r ...); a lo mejor resulta más interesante dar (alpha, beta, TAS, ...) Esto implicaría hacer una función que pasase de esas variables al vector de estado inicial (usando las funciones que ya tenemos en utils. Pero hay que definir qué variables pueden ser interesantes como iniciales y esto no lo sabremos hasta que no nos pongamos a pensarlo.

* Probando el primer caso, se vio que hacía falta un trimador... Si se prueba la pelota no hace falta un trimador. Este debería ser nuestro primer caso de prueba (sin Magnus).

* Del vector de estado interesaría sacar otras muchas variables (otra vez, alpha, beta...) en cada paso de integración, ya que son necesarias, por ejemplo, en el cálculo de las fuerzas para el siguiente instante. De nuevo surgirá la necesidad de hacer funciones que las saquen tirando de las que ya tenemos en utils.

Esta es más o menos la problemática...

**Lo ideal sería ir probando y que vayamos abriendo issues con las sugerencias de nuevas implementaciones que cada uno juzgue necesarias**.

:airplane: :airplane: :airplane: :airplane: :airplane: 

### Aunque no lo creamos, en muy poco tiempo hemos conseguido un simulador que :boom: casi :boom: funciona. El "último esfuerzo" está aquí.

Juanlu001, 2016-08-17:
¿Con los ejemplos actualizados en #65 podemos cerrar esta tarea?

AlexS12, 2016-08-17:
Cerramos!

closed 2016-08-17 by AlexS12