AeroPython/PyFME #38: Altimetría

AlexS12, 2016-01-15:
La altitud que entra en la isa es la geopotencial. Se me ocurre que quizá esté bien tener dentro del paquete utils, un módulo `altimetry.py` que se encargue de conversiones de altitud.

* Altitud geométrica (h): sería la distancia al centro de la tierra menos el radio (medio) de la tierra.
* Altitud barométrica: es la altitud que en la atmósfera estándar corresponde al valor de presión que lee el barómetro del avión en vuelo.
* Altitud geopotencial (H): es la altitud geométrica considerando la gravedad constante. Para obtener la geométrica a la que corresponde: se busca la altitud (geométrica y con gravedad dependiente de la altura) que tiene el mismo potencial que la altitud geopotencial dada. Acaba siendo: h = r_t * H / (r_t - H)

Podemos ir implementando el módulo con conversiones del estilo: `geopotential2geométrical(h)`...

DLpadilla, 2016-01-17:
Yo me quedo con este issue. Si os parece bien.

Juanlu001, 2016-01-17:
@DLpadilla 👌

AlexS12, 2016-01-17:
Estupendo! había pensado en ti!

AlexS12, 2016-02-15:
Se puede cerrar esto con lo que subió ayer Dani! Pull request #55

closed 2016-02-15 by AlexS12