AeroPython/PyFME #43: refactorEulerAnglesCode

LUIS92AERO, 2016-01-17:
He visto el código de Juan sobre coordinates.py y me pareció interesante escribirlo de otra forma. En vez de tener que definir mu, gamma y la X fancy (otra matriz infumable con senos y coseno) como los ángulos de Euler ( ya sea para la actitud de un avión o del vector velocidad) son una composición de rotaciones elementales voy a refactorizar el código y mutliplicar matrices elementales de rotaciones sobre cada eje para todos los cambios entre los ejes (wind, body, horizon).

La forma de Juan es mas rápida pq la multiplicación ya está echa de forma analítica y también tiene tests y todo así que a priori este módulo estaría flotando por si solo. Simplemente escribí este side project para volver a pensar en ejes y probar un par de tests. Quizá en el futuro nos venga bien.

AlexS12, 2016-01-17:
Quizá lo que sí que nos vendría bien es una función que nos permita hacer un giro de magnitud alpha en torno a un eje (x, y, z) que le indiquemos por cabecera. Igual en alguna situacíon es útil.

Definirse todos los cambios llamando recursivamente a esa función es un poco matada si vamos a seguir usando las otras, no?

Juanlu001, 2016-01-21:
¿Queréis decir algo así?

http://poliastro.readthedocs.org/en/latest/api.html#poliastro.util.rotate

AlexS12, 2016-02-15:
Va a avanzar por aquí la cosa algo? O quedó muerto?
@LUIS92AERO, si vas a centrar los esfuerzos en otra cosa lo cerramos, si crees que se hará algo en algún momento, lo dejamos abierto! ;)

LUIS92AERO, 2016-02-16:
Lo voy a cerrar. Este mes estoy a tope. Cuando me ponga con pyfme mejor me centro en modelos aerodinámicos que veo que aportaría mas valor.

closed 2016-02-16 by LUIS92AERO