AeroPython/PyFME #45: TAS CAS EAS

JuanMatSa, 2016-01-18:
Si no va dentro de anemometry, podría encargarme de esto

Juanlu001, 2016-01-18:
Que lo asigne @AlexS12 que es el project manager :stuck_out_tongue:

JuanMatSa, 2016-01-21:
pull request is almost ready.

Just a question: Which input parameters would you select? density and static pressure at flight level, or   sound velocity and static pressure?

At the beginning I chose density and static pressure (cause some changes such as tas2eas just need density)... but, Matlab has a similar function (http://es.mathworks.com/help/aerotbx/ug/correctairspeed.html) and they use sound velocity and static pressure. so... what do you think about?

Additionally, I decided to make 6 different functions: tas2eas, eas2tas, cas2eas, eas2cas, tas2cas, cas2tas instead of one asthe matlab tool. I think it is clearer.

cheers!

JuanMatSa, 2016-01-21:
y ahora digo yo, para qué carajo lo pongo en ingles? espero que se entienda al menos

Juanlu001, 2016-01-21:
Se entiende perfectamente :smile:

Yo creo que dar la densidad es más directo que dar la velocidad del sonido, pero todo depende de cómo se vaya a usar la función.

En cuanto a las 6 funciones, está perfecto - aunque quisiéramos añadir una "gran función" encima de todo eso para que funcionase como MATLAB, internamente tiene sentido tener una función para cada conversión.

_Talk is cheap, show me the code_

commit 49f14509 (2016-01-26): Added tascaseas functions to anemometry module fix #45
closed 2016-01-26 by AlexS12