AeroPython/PyFME #46: Revisión formato docstrings para documentación con Sphinx

AlexS12, 2016-01-18:
Ver:
https://github.com/AeroPython/PyFME/pull/42#issuecomment-172354721
Básicamente, la documentación no se muestra del todo bien debido a algunos errores de formato (espacios, saltos de línea...)
Hay que revisar la documentación y aprovechar la pasada para subsanar otros posibles errores.

JuanMatSa, 2016-01-25:
si, de hecho acabo de ver que en coordinates no coge bien el primer cabecero Parameters. tengo que revisarlo.

Juanlu001, 2016-01-27:
Juan, si te parece me asigno yo esto y le doy una pasada completa a toda la documentación para quitar los errores del Sphinx. Así tú te puedes concentrar en la chicha :wink:

JuanMatSa, 2016-01-27:
perfe!

Juanlu001, 2016-03-22:
Parece que en Sphinx 1.4 van a introducir un modo que solo comprueba la sintaxis http://www.sphinx-doc.org/en/master/changes.html#release-1-4-alpha1-released-feb-14-2016 sin duda ayudará a agilizar un poco este tema, pero sigo pudiendo hacerlo con la versión actual.

Juanlu001, 2016-03-29:
Ya ha salido Sphinx 1.4 http://www.sphinx-doc.org/en/stable/changes.html#release-1-4-released-mar-28-2016

AlexS12, 2016-03-29:
:+1: perfecto!

Juanlu001, 2016-03-31:
¡Faltaba la extensión `napoleon`, que es la que implementa la convención de NumPy! Por eso hay tantos errores. Voy a cambiarlo directamente en master y trabajar desde ahí.

Juanlu001, 2016-03-31:
Referencia: http://www.sphinx-doc.org/en/stable/ext/napoleon.html

Juanlu001, 2016-03-31:
Si os dais cuenta, casi las únicas advertencias que han quedado es por referencias con identificadores duplicados y algunos problemillas en la cabecera de `isa.py`:

http://readthedocs.org/projects/pyfme/builds/3865441/ (último bloque)

Voy a solucionar estos problemas con un pull request.

PR #61 (2016-03-31): Fix docstrings
Closes #46.
closed 2016-03-31 by AlexS12