AeroPython/PyFME #52: Nuevo modelo de avión F-16

AlexS12, 2016-01-31:
En el libro :books: _Stevens, BL and Lewis, FL, "Aircraft Control and Simulation", Wiley-lnterscience_ :books: vienen datos para un modelo de F-16 :boom: . El modelo de avión está hecho en subrutinas de Fortran 70 en el Anexo A y la descripción del modelo aerodinámico en la sección 2.3. La verdad es que estaría chulo implementarlo pero es una tarea gorda que igual requiere de un par de personas trabajando en ello. ¿Qué os parece?

Me parece recordar que @LUIS92AERO estaba interesado en los temas de aerodinámica y no sé si a @olrosales le sigue tirando el tema de los modelos de avión o prefiere hacer otra cosa.

Me gustaría también que nuestro experto en licencias :bowtie:  (y muchas otras cosas), @Juanlu001, nos comentase si sería un problema implementar ese modelo, por el hecho de que ya esté hecho en Fortran 70 en el libro.

Antes de que alguien se ponga a ello, hay algunos temas sobre cómo estructurar el modelo de cada avión que me gustaría comentaros y que decidiésemos entre todos.

Juanlu001, 2016-02-01:
@AlexS12 Con respecto al software de Stevens & Lewis, he buscado en el preámbulo y en los apéndices algún párrafo que aclarase el copyright de las subrutinas pero no he encontrado nada, así que aplicaría lo mismo que al resto del libro:

> Reproduction out translation of any part of this book [...] without the permission of the copyright owner is unlawful.

Por tanto habría que hacer lo que hice yo cuando quise redistribuir el software de Vallado con poliastro: mandar un correo al autor especificando exactamente lo que se va a hacer: ¿una copia tal cual o una "traducción" a otro lenguaje? ¿Bajo qué licencia se va a redistribuir? Y pedir permiso.

AlexS12, 2016-03-23:
@DLpadilla aquí están todos los aviones de la gente de JSBSim, en muchos de ellos viene de dónde han ido sacando los datos.
https://sourceforge.net/p/jsbsim/code/ci/master/tree/aircraft/

Por otro lado está [open falcon](https://github.com/FreeFalcon/freefalcon-central/tree/develop/src/sim) y seguro que si investigamos un poco más vamos sacando cositas.

AlexS12, 2016-03-25:
http://www.zaretto.com/content/simulation-aerodata-downloads

modelos muy completos de f14, f15, b747

aqreed, 2016-08-14:
@AlexS12 me esta costando encontrar blueprints de la Cessna 310!

AlexS12, 2016-08-14:
El único criterio que hemos seguido para elegir el modelo de avión es: el que tenga más datos accesibles. Si tienes alguna alternativa que te hace la vida más fácil con DATCOM adelante. :+1:

AlexS12, 2016-08-14:
para no meter demasiado ruido por sitio distinto, vamos a cerrar este issue y seguimos a hablando por #67 

@DLpadilla contamos contigo para el modelo si sigues con ánimo y tiempo! :smiley:

closed 2016-08-14 by AlexS12