AeroPython/PyFME #54: Funciones para generar entradas de mandos

AlexS12, 2016-02-14:
Hola a todos,

Estaría bien, de cara a poder ir probando casos en los que metamos perturbaciones con los mandos que tuviésemos un nuevo module dentro del paquete `utils`, de nombre algo así como `input_generator.py` (u otro que se os ocurra) que contenga funciones para generar funciones del estilo de un salto escalón, un doblete, una entrada harmónica...

las cabeceras podrían ser algo así como:

`doublet(t_ini, T, A, t, offset=0, var=None)`

t_ini: tiempo inicial; T: Periodo o duración de la perturbación; A: amplitud pico-pico; t: vector de tiempos de la simulación; offset: cte a sumar a la perturbación; var: vector con una perturbación anterior, por ejemplo un seno, que se superpone al doblete.

La salida sería una vector del tamaño de t que da en cada instante el valor de la perturbación.

Cosa a tener en cuenta, pero innecesaria de momento:
Hacer de esto un generador y dejarlo preparado así para que si en algún momento el dt en la simulación va cambiando en función de lo que diga el integrador, se puedan ir genrando los inputs.

¿Alguien se anima?

Harían falta al menos:

* Doblete (se puede construir con dos escalones).
* Escalón
* Entrada harmónica

Otras:
* Entrada en rampa
* Triángulo (se puede construir con dos rampas).
* Parábola
* Exponencial

olrosales, 2016-02-15:
Voy a intentar implementar alguna

AlexS12, 2016-02-15:
Guay! Abre el pull request cuando lo empieces y vamos viendo los avances y te echamos un cable con lo que haga falta.

Por cierto, si por el camino se te ocurre alguna otra forma de hacerlo, con una cabecera distinta o algo así, adelante eh!

olrosales, 2016-02-15:
@AlexS12 Alex, no entiendo la necesidad del parametro var

AlexS12, 2016-02-15:
Lo puse con la idea de que se pudiesen ir sumando inputs usando menos memoria. Si te pasan ya el input con una función senoidal y sobre ella superpones el doblete, no se van acumulando vectores que pueden llegar a ser bastante grandes y que al final se van a sumar en uno.

No obstante, es cierto que si el usuario es avispado puede evitar acumular los vectores él solo. Desde mi punto de vista no hace daño dejarla puesta, pero queda a tu criterio :wink:

olrosales, 2016-02-15:
@AlexS12 En realidad lo preguntaba porque pienso que parece más claro para el que lo vaya a usar ver que una función se llame doublet y dentro haya eso y no una mezcla. Pero es cierto que es utill

AlexS12, 2016-03-23:
Se cierra este issue y continúa la discusión en #56

closed 2016-03-23 by AlexS12