AeroPython/PyFME #57: Tutorial o diagrama sobre como ejecutar casos.

olrosales, 2016-02-16:
Creo que es interesante, crear un breve tutorial o un diagrama con los pasos que ha dado @AlexS12 para ejecutar algunos casos de ejemplo, ya que creo que esto facilitaría que cualquiera entienda el funcionamiento del simulador completo. ¿Qué os parece?

AlexS12, 2016-02-16:
Me parece guay! incluso si alguien mejora un poco el orden del script para alguno de los casos que hay o para uno nuevo, será bien recibido. 

La idea a no tan largo plazo es que muchas de las cosas que se hacen ahí queden encapsuladas en clases y funciones. Si se os ocurre alguna, adelante con la implementación.

AlexS12, 2016-03-25:
@olrosales, he estado mirando y cualquiera de los ejemplos vale. Los arreglé todos un poco antes de hacer el merge. Ahí se puede seguir bien el flujo del script. Cualquier cosa, pregúntame y ya está.

Hay unas cuantas formas de meterle mano a eso para que sea más fácil lanzar casos, ve dándole una vueltecilla cuando tengas tiempo y vemos cual es mejor.

AlexS12, 2016-04-20:
¿Ha avanzado algo esta parte?

Se ha hablado también en varios momentos de hacer algún script que automatice más el lanzamiento del caso. Recuerdo que le pegaste un buen repaso a cómo funcionaba esto @olrosales y sé que @JuanMatSa también ha invertido bastante tiempo.

Tenemos el issue marcado para la v0.1 el 1 de mayo. ¿Qué opción os parece mejor que tomemos: tutorial o script? ¿Cómo lo veis de tiempo?

olrosales, 2016-04-20:
@AlexS12 Alex, yo he estado trabajandolo, y empece la parte del módulo de gráficas, pero esta semana estoy cn la mudanza a la ciudad y los papeles. De verdad lo siento mucho pero hasta 4 de mayo mínimo lo tengo parado

Juanlu001, 2016-07-07:
Esto al final lo partimos en dos: los diagramas para @martosc y el tutorial para @AlexS12. @olrosales si te parece concéntrate en la parte gráfica.

AlexS12, 2016-08-17:
El tutoriales de  cómo lanzar casos, vamos a dejarlo un poco congelado. Con la nueva interfaz los ejemplos son sencillos de entender. Cuando @martosc tenga los diagramas intentaré acompañarlos de una explicación del funcionamiento del simulador. Podemos empezar en la wiki y si va quedando bien, escribirlo en la doc

Juanlu001, 2016-09-04:
¿Qué tal van lo diagramas @martosc? @olrosales, si tenías algo empezado ¿podrías subirlo? Así vamos desbloqueando las cosas que son de documentar :grin:

AlexS12, 2016-11-04:
En esta parte había algo hecho pero que nunca llegó a publicarse, ¿cierto? ¿Sería posible sobir a la wiki al menos lo que sí está hecho este fin de semana?