AeroPython/PyFME #58: Módulo que represente todas las gráficas que consideremos interesantes.

olrosales, 2016-02-17:
¿Qué os parece si dejamos implementado un módulo que represente las soluciones y las entradas? Cualquier usuario puede hacer eso fácilmente pero de esta manera, nos resultará más cómodo para lanzar casos. En caso de que os parezca bien, ¿Qué ponemos?

AlexS12, 2016-02-17:
Me parece muy buena idea. 

Se me ocurren varias formas de hacerlo o mejor dicho, diferentes requisitos. ¿Qué tienes en mente? ¿una función y que pinte todos los estados? una función a la que le pases la línea de tiempo y el número de cosas que quieres pintar y que te saque los subplots? ¿otros? ¿Lo definimos un poco más?

Juanlu001, 2016-07-07:
Asignado a @DLpadilla tal y como hablamos en la reunión del otro día.

olrosales, 2016-07-07:
@Juanlu001 Esto lo tenia yo asignado y empezado

olrosales, 2016-07-07:
@olrosales Pero que lo haga quien quiera

Juanlu001, 2016-07-07:
¡Os podéis coordinar entre los dos! Pero GitHub no me deja asignar tareas a varias personas :sweat_smile:

AlexS12, 2016-07-09:
Si lo tenías empezado, @olrosales , puedes hacer el pull request y así colaboráis ahí para adaptarlo y meterlo en la estructura orientada a objetos. Seguro que se puede aprovechar lo que tienes hecho!

Juanlu001, 2016-08-17:
En 07abcdeb1dc381b110e32ff577b56dcbbe532053 @AlexS12 añadió el código necesario para poder hacer plotting, pero de momento vive en el script de ejemplo y no hay un módulo pensado para ello. @olrosales @DLpadilla si teníais algo empezado antes de integrar #65 supongo que ya no funcionará.

En la carpeta de ejemplos hay un poco de código de plotting que puede servir de inspiración:

https://github.com/AeroPython/PyFME/tree/2d4c4018/examples

Tengo mis dudas sobre si `BatchSimulation` debería tener un método `plot`. La otra cuestión es si queremos tener algo así para la 0.1 o si lo posponemos para la 0.2.

DLpadilla, 2016-08-17:
En los próximos meses voy a estar desconectado si @olrosales  tiene ánimo y ganas que intente sacarlo para la versión 0.1, pero personalmente lo pospondría a la 0.2. Habría que darle una buena pensada de que queremos sacar o visualizar y si podría un módulo a parte el de representación o que queremos que forme parte del `BatchSimulation` .

AlexS12, 2016-08-17:
Voy a pensar estos días en alguna manera más elegante de almacenar y representar los resultados. Si hay aportaciones, id comentando algo por aquí. Si no, ya iré abriendo algún issue con tareas más concretas

Juanlu001, 2016-09-04:
En los ejemplos que @elrond30 está rehaciendo en #72 se ve que hay bastante código repetido en lo que respecta al plotting, si @olrosales quieres intentarlo lo dejamos para esta versión y si no lo pasamos a la siguiente.

Juanlu001, 2016-09-07:
Le voy a quitar el milestone a esto - si entra algo de código para la versión 0.1 perfecto, si no lo discutimos con más calma en la 0.2.

AlexS12, 2016-09-29:
**Contribución Andrés** de #85 (cerrado para que esté aquí todo)

Sería interesante que de output se obtuviera una décima gráfica donde se pueda ver la posición, en grados, de las 3 deflexiones y la posición del throttle también.

aqreed, 2016-09-29:
También sería interesante visualizar L,D,Y en vez de Fx, Fy, Fz...

¿Que pensáis?

AlexS12, 2016-10-03:
Voy a intentar darle una vuelta a la representación gráfica y almacenamiento de resultados la semana que viene si nadie en esta tarea ha podido echarle un vistazo. En cualquier caso, a ver si podemos cerrar la versión 0.1 y empezamos a ver qué hacer con estas cosas.

AlexS12, 2017-11-13:
The graphical output will be managed through pandas DataFrames after merging #104. Even if this drastically simplifies the plotting it would be nice to have a module to manage the plots, and results in general (maybe writing to files, saving simulation sessions...)

aeroaks, 2017-11-14:
Would be interesting to work on this! Are you looking for any particular plotting library? And I suppose interactivity is essential to the output plots !?

AlexS12, 2017-11-14:
Nice! Hi again, Akshay!
I can remember your examples with plotly. They were really nice. Given that we have our results in dataframes now, we have Bokeh and matplotlib “out of the box”. Is it also that direct with pandas and plotly?

Interactivity is important and in the long term, it would be nice to have real time plotting for example. Right now, I think it would be enough to define the interface of some functions to be able to plot properly related groups of variables (we can discuss here which ones) and generate static plots. 

If I had to do it myself right now, I would use matplotlib, but if you start working on this I don’t have any inconvenience to discuss about other approaches or to use other the library that makes you feel more confortable or motivated.

aeroaks, 2017-11-15:
@AlexS12 I am open to using any plotting library. Bokeh is my current favourite for doing general plotting, but there is always matplotlib!

I totally agree with the idea for a plotting/output handling engine which makes it easy to do the plotting, comparison and export of data. For that I would like to also have some understanding of the restructured code.

I prefer to take this up into small tasks, which would help in better feedback ?! If you could define those variables and their locations I would try to look it up.

AlexS12, 2017-11-18:
Regarding the library for plotting I think bokeh is great and might be a sensible decision :ok_hand:. @AeroPython/pyfme does anyone have another suggestion?

Regarding the new code:
- The results during a simulation are stored in dictionary https://github.com/AlexS12/PyFME/blob/461ca9a593b4d74216b3aff650f4fbd09e170633/src/pyfme/simulator.py#L124
- After each propagation, the results are returned as a `pandas.DataFrame`: https://github.com/AlexS12/PyFME/blob/461ca9a593b4d74216b3aff650f4fbd09e170633/src/pyfme/simulator.py#L177
- The variables that are stored and there location in the different simulation objects are listed here: https://github.com/AlexS12/PyFME/blob/461ca9a593b4d74216b3aff650f4fbd09e170633/src/pyfme/simulator.py#L42
- The user can customize that in a pretty uncomfortable way right now which consists in providing an alternative dictionary like the one above when initializing a simulation: https://github.com/AlexS12/PyFME/blob/461ca9a593b4d74216b3aff650f4fbd09e170633/src/pyfme/simulator.py#L94

I have some ideas on how to organize this, but for sure it will be enriching to hear yours!

Juanlu001, 2017-11-18:
+1 for Bokeh :)