AeroPython/PyFME #59: Guía para poder contribuir a ramas de otras personas.

AlexS12, 2016-03-23:
Como hemos hablado en la reunión, a veces es necesario tener los cambios que otra persona ha hecho en una rama propia y que está todavía en proceso de pull request; unas veces para contribuir al pull request y otras para incorporar funcionalidades nuevas que requieren de otras que no están en la rama master todavía.

Me quedo encargado de hacer una entrada  detallada en la wiki que explique el proceso.

@Juanlu001, ¿qué opinas del siguiente proceso?

Para contribuir a un pull request:
* Añadir al `remote` el fork del usuario que ha iniciado el pull request y hacer `fetch`.
* Crear una rama propia partiendo de la rama a la que se quiere contribuir.
* Hacer cambios, hacer push al repositorio remoto propio.
* Hacer pull request a la rama del usuario que hizo el pull request inicial `User/PyFME/rama`.

Para iniciar una rama nueva partiendo de una que está en proceso de revisión y no se ha integrado en el master (aunque pueda ser peligroso):
* Añadir al `remote` el fork del usuario que ha iniciado el pull request y hacer `fetch`.
* Crear una rama propia partiendo de la rama que está en proceso de revisión.
* Funcionamiento normal haciendo pull request a `AeroPyton/PyFME`.

Juanlu001, 2016-03-23:
¡Me parece perfecto el proceso! Nada que añadir :+1:

AlexS12, 2016-03-23:
Después de comer hago una entrada tomando como ejemplo una contribución a la rama de @olrosales #56 en la que cree un test y le haga pull request para que lo incluya.

AlexS12, 2016-03-23:
https://github.com/AeroPython/PyFME/wiki/Tutorial-paso-a-paso:-contribuyendo-a-un-pull-request

¡Ahí está la guía! Si algo no se entiende, me lo comentáis. Si creéis que algo puede explicarse mejor o más detalladamente, modificadlo a vuestro antojo.

@olrosales tienes un pull request que tienes que aceptar si te parecen bien los cambios que he hecho :hand: https://github.com/olrosales/PyFME/pull/1 Una vez que aceptes, esos cambios quedarían dentro de la rama con la que tú estás haciendo pull request https://github.com/AeroPython/PyFME/pull/56

closed 2016-03-23 by AlexS12