AeroPython/PyFME #62: Revision del trimmer

JuanMatSa, 2016-04-01:


AlexS12, 2016-04-20:
Hola @JuanMatSa, me consta que has estado echándole un vistazo al trimmer y que tenías algunas cosas que preguntarme y alguna sugerencia. Lo dejo por aquí para que nos acordemos de que esto no está cerrado

Juanlu001, 2016-07-07:
En la reunión del otro día acordamos que yo me encargaría de simplificar la API del trimmer. ¿Queda algo desde el punto de vista matemático que @JuanMatSa quiera aportar? @AlexS12 ¿podemos cerrar #63 como duplicado de esto o son dos cosas diferentes?

AlexS12, 2016-07-09:
@JuanMatSa tenía alguna duda sobre el algoritmo que teníamos que comentar. Otro aspecto de mejora era la inicialización, sobre la que habíamos quedado en pensar. No obstante, creo que puede quedar para más adelante.

AlexS12, 2016-08-17:
El código del trimador ha ido cambiando, pero, en esencia, el método de trimado es el mismo. @JuanMatSa, si quieres revisarlo cuando puedas, adelante, seguro que aún se puede pulir y tú estabas bastante puesto en este tema.

Por mi parte, me pondré con los tests pronto para tenerlos listos para la  v0.1. Se agradece colaboración, si hay algún interesado, que se manifieste :ghost:

Cosas por hacer:

 - [x] tests de funciones de ligadura:
   - [x] `turn_coord_cons`
   - [x] `turn_coord_cons_horizontal_and_small_beta`
   - [x] `rate_of_climb_cons`

- [ ] __tests del trimador para cessna 310__: comprobaría que además de dar un residuo de los mínimos cuadrados bajo:
  - [ ] matiene condición de vuelo est, rec, uni, alas a nivel después de n segundos
  - [ ] asciende los metros correspondientes para el gamma dado sin desviarse lateralmente después de n segundos
  - [ ] gira los grados correspondientes para el turn rate dado sin descender ni ascender después de n segundos

PD. tengo curiosidad por ver cómo funciona esto con el nuevo modelo de avión #67 :yum:

AlexS12, 2016-11-04:
Por lo que veo aquí nos quedarían por implementar los tests relativos al trimmado de alguno de los aviones. @JuanMatSa te animas al finalo andas pillado de tiempo. La intención es poder cerrar esta versión este fin de semana.

closed 2017-11-13 by AlexS12