AeroPython/PyFME #64: Bienvenida a curiosos e interesados en colaborar.

AlexS12, 2016-04-26:
Sería una buena idea hacer una entrada en la wiki que presentara el proyecto a gente que pasa por aquí. Para nosotros es más o menos obvio, pero para alguien que viene de pasada, todo es un misterio. Creo que lo estamos haciendo bastante bien con la wiki, así que puestos a mejorar... 

La entrada debería contar:

- qué hace el simulador hasta ahora.

- qué está previsto que haga próximamente y a largo plazo.

- qué tipo de tareas necesitamos llevar a cabo desde un punto de vista general (para el detalle ya tenemos los issues).

- email de contacto o similar.

- link al flujo de trabajo.

Podemos poner el link a esa página en el readme o directamente, todo ahí.

¿Se os ocurren más cosas? ¿Alguien se anima a ir poniendo algo en la wiki?

Juanlu001, 2016-04-26:
@AlexS12 totalmente, creo que ayudaría a la gente a colaborar

Juanlu001, 2016-07-07:
Asignando esto a la versión 0.1.

AlexS12, 2016-07-09:
Algún voluntario para darle un empujoncillo a https://github.com/AeroPython/PyFME/wiki y yo me ofrezco a seguir completando

AlexS12, 2016-08-08:
Se puede seguir completando https://github.com/AeroPython/PyFME/wiki y @martosc está encargado de la página https://github.com/AeroPython/PyFME/wiki/%C2%BFQu%C3%A9-necesito-para-empezar%3F

AlexS12, 2016-08-14:
Veo que @martosc subió hace unos días cosas a [esta página](https://github.com/AeroPython/PyFME/wiki/%C2%BFQu%C3%A9-necesito-para-empezar%3F). No sé si lo da por terminado.

En cualquier caso, aprovechando el tirón del otro día, os animo a todos (@AeroPython/pyfme) a echarle un vistazo y añadir lo que creáis necesario (problemas particulares que tuvisteis, cosas que creáis que faltan...).
Podéis echarle también un vistazo a https://github.com/AeroPython/PyFME/wiki y aportar lo que os parezca conveniente.

Dejo esto abierto un par de días más para comentarios y cerramos el issue. :tada:

Juanlu001, 2016-09-04:
Lo que subió @martosc tiene buena pinta y parece que va en la dirección correcta, @AeroPython/pyfme ¿os animáis a continuarlo? Que @AlexS12 ya tiene bastante con #70 :stuck_out_tongue:

AlexS12, 2016-09-05:
+1 ¿alguien se anima?

aqreed, 2016-09-11:
@AlexS12 aquí no me entero muy bien de cual sería la forma de colaborar

AlexS12, 2016-09-11:
@aqreed , la forma de colaborar está explicada en: https://github.com/AeroPython/PyFME/wiki/Tutorial-paso-a-paso-del-flujo-de-trabajo. Lo que subió @martosc tiene que ver con lo que es necesario instalarse para poder trabajar. Quedarían por hacer los diagramas de los que se habla en #57 explicando un poco cómo funciona el simulador, por si quieres echar un cable a @martosc.

He marcado en https://github.com/AeroPython/PyFME/issues/64#issue-151042671 lo que está incluido y no. En los primeros puntos sería dar una idea muy general. Si te hace, todo tuyo!

AlexS12, 2016-11-04:
Veo que no hay nada nuevo en  https://github.com/AeroPython/PyFME/wiki. 
¿@martosc, @aqreed, vais a poder completar alguno de los puntos marcados en   https://github.com/AeroPython/PyFME/issues/64#issue-151042671?

aqreed, 2016-11-06:
@AlexS12 ya he hecho algo, que puedes ir mirando. Estoy redactando qué hace y qué queremos que haga, en esto último supongo que hay que comentar lo de conectarlo a flight gear no?

AlexS12, 2016-11-06:
Efectivamente, lo de Flight Gear, las cosas que has ido sugiriendo de modelizar actuadores, modelo más completo (poder modificar centrado, momento de inercia o tener en cuenta el consumo de combustible), integrar otros sistemas dinámicos (cuaterniones, sistema inercial teniendo en cuenta el giro terrestre para otro tipo de aplicaciones), autopilotos y sistemas de aumento de la establidad.

Algunos están muy lejos y habrá que decidir qué camino tomamos, pero no hace daño ir sugiriendo y que quede ahí puesto