AeroPython/PyFME #66: Actualizar precisión de los tests

Juanlu001, 2016-08-09:
Hemos visto que algunas precisiones no están bien ajustadas a los resultados, vamos a echar un vistazo a ver si se puede afinar un poco.

Juanlu001, 2016-08-17:
@miqlar Como todos los tests de la atmósfera (`src/pyfme/environment/tests/test_isa.py`) se han roto por el cambio de interfaz, puedes probar a arreglarlos. ¡Si tienes dudas avísanos!

AlexS12, 2016-08-17:
Vamos a meter en este lote también `test_anemometry_stagnationpressure` son dos tests y el problema en las ligeras diferencias en los resultados puede tener que ver con alguna constante (estilo velocidad del sonido, densidad atmosférica a nivel del mar... que haya cambiado de valor).

  - [ ] `test_isa.py`
  - [ ] `test_anemometry_stagnationpressure.py`
  - [ ] `test_anemometry_tascaseas.py`

no necesariamente tiene que hacerlos todos @miqlar, que él empiece por la isa, si quiere, y si a alguien más le apetece, es un buen sitio para empezar!

Id comentando por aquí para no pisarnos.

Juanlu001, 2016-08-17:
Me parece recordar que @martosc quería mirar los tests de velocidades. Ahora se pueden asignar las tareas a varias personas así que... :muscle:

miqlar, 2016-08-24:
¿De donde habéis sacado los valores de la velocidad del sonido que usáis en test_isa.py?

Tienen precisión 4 decimales y en http://ntrs.nasa.gov/archive/nasa/casi.ntrs.nasa.gov/19770009539.pdf (la fuente) solo 2. 

Por entender de donde salen, porque los tengo que cambiar todos en principio (no son correctos)

Juanlu001, 2016-08-24:
2016-08-24 11:37 GMT+02:00 Miquel Larsson <notifications@github.com>:

> ¿De donde habéis sacado los valores de la velocidad del sonido que usáis
> en test_isa.py?
>
> Tienen precisión 4 decimales y en http://ntrs.nasa.gov/archive/
> nasa/casi.ntrs.nasa.gov/19770009539.pdf (la fuente) solo 2.
>
> Por entender de donde salen, porque los tengo que cambiar todos en
> principio (no son correctos)
>

Esos tests se introdujeron aquí:

https://github.com/AeroPython/PyFME/commit/a8bb99b2ebc18e8dcdf8d1891573127f00381179

(lo he visto en
https://github.com/AeroPython/PyFME/blame/2b47a07/src/pyfme/environment/tests/test_isa.py#L28
)

Y en su día parece que los tests pasaban:

https://github.com/AeroPython/PyFME/pull/44

@martosc ¿Puedes arrojar un poco de luz sobre el tema?

miqlar, 2016-08-24:
La cosa es que el test mas simple (nivel del mar) espera:

` expected_a = 340.4155  # m / s`

Lo que no está bien, porque en el documento fuente como hasta en el propio archivo de constants está definido:

`SOUND_VEL_0 = 340.293990543  # sound speed at sea level (m/s)`

La razón parece ser que en este file de constants la gamma está definida como

`GAMMA_AIR = 1.401`

Lo que (revisando documentación) no está bien. Cambiando al valor correcto 1.4, ya se obtiene el valor de la velocidad del sonido correcto. Con esto, hay que cambiar todos los valores de a's que comprueba el test a los valores correctos.

Juanlu001, 2016-08-24:
Ese gamma 1.401 tiene mala pinta, y además se está sobreescribiendo su valor en algunos sitios:

https://github.com/AeroPython/PyFME/blob/5fe928e/src/pyfme/utils/tests/test_anemometry_tascaseas.py#L21

¿Podría esto tener algo que ver con los fallos?

Juanlu001, 2016-08-25:
Arreglado en #74 :tada:

closed 2016-08-25 by Juanlu001