AeroPython/PyFME #67: Investigar modelo aerodinámico más complejo usando DATCOM

Juanlu001, 2016-08-09:
Dad la bienvenida a Lord DATCOM :laughing:

aqreed, 2016-08-14:
Bueno, el nuevo modelo obteniendo los coeficientes con DATCOM y AVL quedaría algo así:

        # |CL| = |CL_a  CL_de   CL_a_d  CL_q| |    alpha     |
        # |CD| = |CD_a  CD_de   0       0   | |delta_elevator|
        # |Cm| = |Cm_a  Cm_de   Cm_a_d  Cm_q| |(c/2V)*alpha_d|
        #                                     |   (c/2V)*q   |
        #
        # |CY| = |CY_b  CY_da   CY_dr   CY_p   CY_r| |    beta     |
        # |Cl| = |Cl_b  Cl_da   Cl_dr   Cl_p   Cl_r| |delta_aileron|
        # |Cn| = |Cn_b  Cn_da   Cn_dr   Cn_p   Cn_r| |delta_rudder |
        #                                            |  (b/2V)*p   | 
        #                                            |  (b/2V)*r   |

- DATCOM devuelve los C* _a  y los C*_b como una tabla, con los valores para alpha=0 incluidos
- en primera aproximación (primeros tests) los valores de las amortiguaciones (los C*_{p,q,r y alpha_d}) podemos tomarlos por nulos, pero para el modelo final son bastante importantes
- el programa ya tiene en cuenta la incidencia del empenaje horizontal

DUDA: en qué modelo habéis visto que se usa Dbeta_Dt = 0  # Rate of change of AOS ?

AlexS12, 2016-08-14:
Hola Andrés, bienvendio por aquí también. Te veo ya a tope y eso es estupendo!

Entiendo, por lo que dices arriba, que de DATCOM sólo salen coeficientes con respecto a beta y alpha, no? De dónde sacaríamos el resto?

El tema de las derviadas de control, no me parece problemático. La ventaja que le veo a tener el coeficiente como tal es que se puede utilizar para alguna aproximación teórica de validación. En cualquier caso, siempre se puede sacar dividiendo entre el delta de mando que le corresponde.

Creo recordar que no hemos encontrado ningún modelo que incorpore "beta punto" pero lo dejamos puesto por darle más generalidad.

aqreed, 2016-08-14:
DATCOM da coeficientes en función de h, Mach, alpha y las diferentes deflexiones, según cada caso. Para una avioneta como la que estamos pensando (Cessna) h y Mach no tienen casi influencia, se pueden obviar en primera aproximación. No hacer el barrido en beta es una de las limitaciones del programa, el sentido común dice que no es lo mismo en cuanto a efectividad deflectar el timón de dirección con beta=0º que con beta=10º (Cn_beta debería depender de beta también), pero es un software que se lleva usando mucho tiempo para cosas como esta y nadie ha muerto aún!

Si hay deflexión por medio, no devuelve la derivada del coeficiente en sí, sino el incremento del coeficiente debido a esa deflexión. Por ejemplo en el caso del coeficiente de resistencia CD, su aumento al deflectar elevator se da como D(CDI) = |CD_delta_elev*delta_elev| en función de alpha y delta_elev. No son evoluciones lineales, ni en delta_elev ni en alpha, sino más bien una superficie irregular con forma de silla de montar, si pintamos alpha y delta_elev en OX, OY y en el OZ el incremento.

Ponerse a dividir para sacar la CD_delta_elev para cada condición de vuelo puede ser poco útil para los primeros modelos. Luego después, si podría ser interesante para lo que hablamos del análisis de estabilidad...

aqreed, 2016-08-16:
Los coeficientes de fuerzas y momentos y sus derivadas ya están calculados con DATCOM, para la Cessna 172:
https://app.box.com/s/qc7fr2121xzpczu4tl5dh4vwihbhirt1
- había datos importantes que faltaban para la 310, de ahí el cambio. En concreto, XYZ del c.g. y los momentos de inercia. Para la 172 los he encontrado aquí: http://doc.gnu-darwin.org/cessna172/linear.html
- se han obtenido para h = 1500m, M = 0.16 y una docena de valores de alpha entre -7,5º y 19.5º, cuando entra en pérdida
- los recorridos de las superficies los he sacado de aquí: http://www.airweb.faa.gov/Regulatory_and_Guidance_Library/rgMakeModel.nsf/0/724e90061c5bf3b1862576260063e599/$FILE/3A12.pdf

Hay algunas cosas que aún no cuadran y que investigaré más a fondo: 
- la pendiente de Cyp me suena que debe ser positiva, aunque para el rango de alpha elegido no creará problemas
- Cnbeta tiene que ser positivo si o si, lo tomaremos como constante Cnbeta = 0.065 mientras tanto

Queda por calcular CYr con AVL, las otras derivadas que "derivan" de CYr (Cl_delta_r y Cn_delta_r) y CY_delta_r.

AlexS12, 2016-08-16:
Estupendo, Andrés! le he echado un primer vistazo rápido y es la leche: ¡¡ahora tenemos un montón de datos!! qué bien! :grinning:  ¿Te ves con ganas de explicar el proceso en la wiki: instalar datcom, generar fichero de entrada, ejecutarlo, estructura del fichero de salida?

De cara a ver cómo empaquetar toda esta info, ¿cuáles son archivos de salida de DATCOM tal cual y cuáles has creado tú? 

A la hora de implementarlo hay varias opciones:

- Directamente en código:
  * ventaja: forma más directa.
  * desventaja, cualquier cambio en el modelo implica repetir todo el proceso y transcribir los cambios en los coeficientes.
- Generarnos unos cuantos .dat más o menos estándar para nosotros con información de diferentes coeficientes y que se carguen al instanciar el avión.
  * ventaja: más fácil cambiar cosas que directamente en código.
  * desventaja: más laborioso y sigue requiriendo intervención manual, pero se pueden hacer scripts que generen estos archivos a partir de la salida de DATCOM.

Por último, Andrés: ¿tienes tiempo e interés por implementar el avión en la clase (podemos quedar un día si lo ves necesario para que te eche una mano) o prefieres parar aquí y que otra persona se encargue?

Con esto damos un paso importante! :rocket:

Juanlu001, 2016-08-16:
Con respecto a lo de DATCOM:

> se pueden generar scripts que generen estos archivos a partir de la salida de DATCOM

@aqreed @AlexS12 ¿Estaría bien tener un PyDatcom? Existe ya un proyecto pero según vimos en el sprint del otro día solo funciona con Python 2 y la documentación es escasa. Algo así merecería su propio repositorio aparte claro. Ya hemos forkeado el original por si puede servir de punto de partida:

https://github.com/AeroPython/pydatcom

Si DATCOM se controla desde línea de comandos, @AunSiro tiene experiencia al respecto y nos podría echar una mano.

AlexS12, 2016-08-16:
La verdad es que si nos vemos fuertes y con ganas, ¿por qué no? ¡si hay voluntarios motivados que vayan levantando la mano! @AeroPython

Lo cierto es que para este milestone quizá queda lejos, no? ¿Cuál creéis que sería la mejor manera de implementar este modelo?

aqreed, 2016-08-17:
Respondiendo a las cosas que planteáis:
- Voy a centrarme en sacar el resto de derivadas de la Cessna 172, necesitaré unos días. Después organizaré la carpeta de box para que sea entendible.
- Sobre hacer la wiki, para ahora sería mucho trabajo, aunque si se pudiera [insertar un pdf](http://stackoverflow.com/questions/25328135/embed-a-pdf-in-github-gitlab-wiki) podríamos poner el capítulo de mi pfc dedicado a eso en detalle.
- Sobre implementar el script para leer los datcom.out y las salidas de avl, me parece lo lógico, pero como tarea secundaria, quizá para después de Octubre. El script ejecutaría los dos programas y devolvería un .dat estandarizado por nosotros y legible desde la clase. Para la Cessna 172 propongo implementar directamente en código por ser más sencillo, y sí, contad conmigo para esta tarea! 

@AlexS12 sí, lo más seguro es que necesite quedar alguna tarde para enterarme mejor del tema de los tests, acuérdate que en la reunión de la semana pasada no pude instalar PyCharm. El problema es que tenía instalado OpenJDK...

Juanlu001, 2016-09-07:
¿Qué tal vas con esto @aqreed? Si tienes ya algo de código puedes abrir un pull request. Así evaluamos en qué estado está y vemos si lo metemos en esta versión, que estaría muy bien.

AlexS12, 2016-09-13:
Como esto está ya casi para terminarse en #79, cerramos, queda aquí la referencia y terminamos de discutir estas cosas en el pull request.

aqreed, 2016-09-14:
Bueno yo lo doy por cerrado si!

Juanlu001, 2016-09-14:
No impacientarse, cuando integremos el pull request esto se cerrará solo :wink:

PR #79 (2016-11-04): cessna172 with [LDY] in wind axes
After indentation modifications, addressing of the remaining FIXMEs and including pqr in the model along with the aerodynamics data.

Closes #67
closed 2016-11-04 by AlexS12