AeroPython/PyFME #68: BUG: discontinuidad ángulos de euler

AlexS12, 2016-08-14:
En `pyfme/utils/coordinates.py`, la función `check_theta_phi_psi_range(theta, phi, psi)` comprueba que el valor de los ángulos de Euler está dentro de rango. Sin embargo, al integrar si el avión tiene orientación `psi=0.05` y en el siguiente instante `psi=-0.05` se lanza un `ValueError` porque `psi` está fuera de rango. Esto fue una idea de @JuanMatSa y mía al implementar los cambios de coordenadas, pero está haciendo que se interrumpa la integración cuando el avión pasa de por este valor de `psi`.

Existen varias opciones para arreglarlo:

1. **Tras cada paso de integración el valor de los ángulos de Euler se mete dentro de rango**: `psi=-0.01` --> `psi=2pi - 0.01`
2. **Eliminar comprobación de rango**. En mi opinión no es buena idea: si no se tienen en cuenta las restricciones en el dominio de los ángulos de Euler a una misma posición del avión le pueden corresponder varia combinaciones de ángulos (por ejemplo: psi=180, theta=60, phi=180, sería igual que psi=0, theta=120, phi=0, sólo la primera es válida).
3. Otras que podáis proponer.

Hasta que se haga merge de #65 el error se puede reproducir poniendo en `oop_example.py` `turn_rate != 0.0  # rad/s` e integrando durante tiempo suficiente.

aqreed, 2016-08-14:
@AlexS12 no entiendo mucho este bug, porque a mi entender el rango en sí es sólo a efectos de entendimiento y visualización. Al menos en mi modelo en su momento yo le dejaba llegar a los valores que tuviera que llegar, y luego hacía el cambio al rango "entendible por el ser humano" cuando iba a representarlo en gráficas.

A los algoritmos en sí les da igual si estamos variando entre 179-->181 o entre 539-->541 (si las ecuaciones no tienen nada raro de por medio, que no lo he mirado).

AlexS12, 2016-08-14:
Efectivamente, los ángulos e las ecuaciones pueden tomar cualquier valor porque van dentro de senos y cosenos. Pero como los ángulos de euler tienen un dominio definido, cuando implementamos los cambios de sistema de referencia decidimos comprobar que se estaba respetando. Es ese cambio el que saltaba.

AlexS12, 2016-08-17:
Finalmente esto quedó resuelto en #65 (está referenciado el código un poco más arriba)
Qué ganas de tener los cuaterniones y jugar un poco a ver qué pasa con este tipo de cosas!  @JuanMatSa

closed 2016-08-17 by AlexS12