AeroPython/PyFME #71: Convertir ejemplos a notebooks de Jupyter

Juanlu001, 2016-08-18:
Ahora mismo los ejemplos están en forma de scripts:

https://github.com/AeroPython/PyFME/tree/2d4c4018/examples

Estaría bien tenerlos en formato notebook, como mínimo por tener las gráficas visibles. Además se podrían crear galerías de ejemplos como esta:

http://mybinder.org/repo/poliastro/poliastro

La tarea sería partir el código en celdas de forma ordenada y dejar las gráficas visibles al final. Puntos extra si incluimos algún widget interactivo :smile:

elrond30, 2016-08-22:
En mi fork del proyecto, he creado los notebooks. La galeria está directamente en examples, y los ejemplos en notebook estan en examples/examples-notebook. No he hecho gran cosa todavía, más que copiar el código, sin ni si quiera ejecutar (tengo algunos problemillas que estoy intentando solucionar). Pero si me gustaría que fueráis echando un vistazo, para ver si van por la ruta adecuada, o hay que cambiar algo, ahora que todavía no está muy avanzada esta parte.

Un saludo

Juanlu001, 2016-09-07:
Esto ya se hizo en #72, ¡cerrando!

closed 2016-09-07 by Juanlu001