AeroPython/PyFME #75: Investigar modelo de motor

Juanlu001, 2016-08-25:
@martosc está investigando un modelo de motor más complejo, que involucre parámetros de control y tomando las ecuaciones reales.

Juanlu001, 2016-09-07:
Este es un cambio bastante trascendente que es mejor que discutamos con calma, lo pasamos a la versión 0.2.

aqreed, 2016-09-11:
Mientras tanto propongo hacer el modelo sencillito de controlar T mediante Omega, como si cambios en la palanca de gases correspondiesen de manera lineal con variaciones de rpm.

T = (4/pi**2) * rho * (omega**2) * (R**2) * C_T

donde C_T se obtiene con J mediante los datos obtenidos en JavaProp que adjunto [aquí](https://app.box.com/s/fmptp3h8zhw8pulfxi68yu07ilt6ol5b). Los datos del motor los he sacado de [aquí](http://www.sensenich.com/files/documents/Specification_Sheets_TC_Data_Sheet_P886_74DM,_74DC,_74DR_1263327040.pdf).

aqreed, 2016-09-12:
Dejo aquí el dato de los valores máximos y mínimos de rpm para el Lycoming O-360, sacados de [aquí ](http://www.yukonflying.com/Documents/Lyc%200perating%20Manual.pdf): 1000 (ralentí) y 2800.

aqreed, 2016-09-13:
Para validar el modelo podemos usar los datos de [este pdf](http://temporal.com.au/c172.pdf) que he encontrado, donde para una V ~ 62 m/s a nivel del mar predice un L/D ~ 6.6, luego para vuelo nivelado T/W = 1 / (L/D) ~1/6.6, siendo W ~1100kg queda que tiene que ser T ~ 1600 N de tracción. Espero no haberme colado en nada.

AlexS12, 2016-09-13:
Estupendo, Andrés. Con esto ya tenemos un modelo decente. A ver cuando podemos incluirlo!

En cualquier caso, tengo en mente que @martosc buscará algún otro modelo más refinado, que a lo mejor podemos generalizar a otros aviones. ¿Es así?

aqreed, 2016-09-13:
Bueno este modelo es el modelo que siempre se usa para aviones de hélices, las ecuaciones son las que derivan de la Teoría de Elemento de Pala. Donde habría que trabajar sería en la forma en que el throttle controla a las revoluciones, lo cual depende de características motoras y no propulsivas. Ya es generalizable a otros motores simplemente cambiando las rpm mínimas y máximas. Cuando solucione el tema del cambio de ejes de la #79 hago un nuevo pull con el modelo para que lo podáis ver.

AlexS12, 2017-11-13:
@martosc are you still interested on developing a more complex and complete model for the engine?