AeroPython/PyFME #82: Referencias para las constantes

AlexS12, 2016-09-18:
Al hilo del último pull request (#81) ha surgido la duda de si poner las referencias de todas las constantes que tenemos disponibles en `models/constants`. Parece una idea sensata y llevadera, ahora que todavía no hay muchas cosas ahí. Así que, generamos la tarea. Puedes ser tú mismo @miqlar, si te apetece, quien la haga. Te la asigno en principio y nos dices.

miqlar, 2016-09-18:
He añadido unos cuantos ya en #81 , así que puedo completarlos sin problema.

Un tema que me tiene confundido es el de la velocidad del sonido con h=0. Estaba definido en constants como **a=340.293990543**, y no encuentro de donde sale. Porque aplicando la fórmula típica de sqrt(gamma*R*T_0), sale **340.293988**.

Diferente y con muchos menos decimales. El US1976 standard solo te define 2.

El tema es que si se cambia, fallan los tests de anemometry.... los tas2eas etc Y a altura h=0 no hay duda en lo que debería salir. Parece como si a se hubiera ido modificando a mano hasta que los tests pasaran. 

Lo revisaré!! Pero alguien sabe de donde sale esa **a** con tantos decimales?

AlexS12, 2016-09-18:
mmmm... me temo que eso está relacionado con el `gamma` (¿era el `gamma`, verdad?) que estaba mal. Seguro que lo calculé con el valor erróneo y claro...

Si en el `US1976 `está con dos decimales, yo dejaría ese valor y cambiaría con tranquilidad los tests de `anemometry`. ¿Qué opinas?

closed 2017-11-13 by AlexS12