AeroPython/PyFME #84: Cambiar la definición de los offsets del input_generator

aqreed, 2016-09-28:
Quedan añadidos a la función (sea ramp, doublet...) durante toda la simulación cuando su utilidad se ciñe al intervalo t = (t_init, t_init + T).

Propongo cambiar las funciones para limitar la suma del offset al intervalo de interés.

AlexS12, 2016-09-28:
A mí en su momento me pareció guay la otra forma, la verdad es que está tampoco está mal. Habría que pensar bien qué criterio tomamos, la verdad es que los dos pueden ser útiles, pero hacen cosas distintas. Me gustaría escuchar alguna opinión más.

Desde mi punto de vista tiene más utilidad sumar el offset solo en el tramo donde se aplique el doblete, rampa... Ya que para sumarlo a toda la señal se puede hacer directamente `signal + offset`. Sin embargo, hay que tener en cuenta que hacerlo en un único tramo puede meter saltos escalón además de la perturbación.

¿qué os parece?

aqreed, 2016-10-02:
También es necesario que la señal se sume, que sea una adición, porque ahora mismo lo que hace es borrar la anterior. 

Esto evita que se pueda sumar un step a una rampa, o un doublet al trimado inicial, por ejemplo.

AlexS12, 2016-10-02:
@aqreed Para que no se sume hay que pasar la señal anterior en var 😉

aqreed, 2016-10-02:
Ah, y una cosa curiosa, no sé porqué pero es imposible definir una amplitud = 0, siempre queda un valor ¿qué pensáis que puede ser?

aqreed, 2016-10-02:
sí, y es un sistema lioso (lo de pasar la anterior en var). Creo que se podría definir mejor.

olrosales, 2016-10-02:
@aqreed en su momento a @AlexS12 (si no recuerdo mal) y a mi nos pareció una buena forma de tener control sobre las señales

olrosales, 2016-10-02:
@olrosales Quería decir que lo modifiqueis como mejor veáis, pero que creo que ya nos pusimos de acuerdo @AlexS12  y yo. Pero como más cómodo sea

AlexS12, 2016-10-03:
A ver, respecto al tema de pasar la señal en `var`, en su momento lo vimos como la manera más sencilla de hacer lo que necesitábamos. Si ahora vemos que es un sistema lioso, siempre hay posibilidad de cambiarlo por algo más cómodo o que nos dé las funcionalidades que buscamos. No hay nada congelado e inamovible.

Se me ocurre que se podría empaquetar la señal en una clase que tuviese los métodos, `doublet`, `step`... pero creo que es añadir un grado más de complejidad. Si se nos ocurren más cosas que necesitemos hacer con la señal igual es un buen precio a pagar.

Respecto de la amplitud cero, aunque sea un caso un poco excepcional, no veo en principio ningún motivo para que no salga bien. La mejor forma de abordarlo creo que es hacer **en una nueva rama y nuevo pull request** un test que no pase ahora mismo y ver magrear el código hasta que pase. Ya que has detectado el problema, @aqreed, ¿puedes hacer el test? si no, puedes pegar el código aquí y seguro que alguien más se anima.

aqreed, 2016-10-04:
Ok, he creado una nueva rama donde modifico el tema de los offsets. El `var `no lo he tocado, y lo de no poder conseguir amplitud cero olvidadlo que ya está solucionado. Mi propuesta es esta: 7798de9 @AlexS12 @olrosales 

Mi intención es poder añadir perturbaciones (sea `ramp`, `doublet `o `step`) al avión trimado de manera más limpia posible.

closed 2016-11-12 by aqreed