AeroPython/PyFME #88: Missing abstract/virtual methods

sanguinariojoe, 2016-11-07:
That's maybe because my C++ deformation...

It seems that `calculate_forces_and_moments` is a common method of all the aircrafts (called in fact from simulator.py). However such virtual/abstract method is missing at base class `Aircraft`.

In Python that's not much relevant in practice, but is veeeeery convenient to can document such function in general.

AlexS12, 2016-11-07:
Well I have to say that, that part of code needs serious review. This class structure is just a first attempt, and in fact, many things from `Aircraft` should be placed somewhere outside, for example, an `Aerodynamics` class (or maybe named tuple...) where `alpha`, `beta` and that kind of attributes could be stored.

Maybe you, who are more familiar with oop could help us designing the structure :wink:

sanguinariojoe, 2016-11-09:
Sure

But how do you want to do that... Because making a fork, and editing the code while you are intensively working on that, sounds like we are having hard conflicts

AlexS12, 2016-11-10:
Yes, we should meet and talk about it first!

sanguinariojoe, 2016-11-20:
OK, I suggest something like that:

[image]

Where everything in the aircraft, including the aircraft itself, is a component. In this way you still have the possibility of setting up simple aircrafts by a class inherited from `Aircraft` and `Wing` at the same time, as well as more complex aircraft structures (in case you have data enough).