AeroPython/PyFME #90: Binder Example Gallery Failure

Strangemother, 2016-11-07:
The example site does not build

http://mybinder.org/repo/AeroPython/PyFME

After roughly 1 minute the interface displays 'failure'

The browser is successfully making requests. The first 60~ requests:

`{"id":"80c376f577cb638e311ce41440af2f21","template-name":"aeropython-pyfme","location":null,"status":"deploying"}`

Followed by a failure response:

`{"id":"b6736adf65897e3cf7e6c241b9e50348","template-name":"aeropython-pyfme","location":null,"status":"failed"}`

Juanlu001, 2016-11-07:
@Strangemother Apparently there are some problems with mybinder:

salgozino, 2016-11-08:
Mybinder is running now, and the example gallery open the Notebook. The problem now is the links to the examples. That links gives an 404 error. You change the adress to the files or something?

aeroaks, 2016-11-08:
it has the same error as the one fixed in #90 . I do not know how mybinder works! do we need to push the changes?

Juanlu001, 2016-11-08:
I just tried http://mybinder.org/repo/aeropython/pyfme and the index.ipynb and its links work as expected, after @aeroaks fixed them. Would you please try again?

aeroaks, 2016-11-08:
It works for me as well! If confirmed by other members, we can close the issue?

Juanlu001, 2016-11-08:
If someone cares to revert 58708c69, we can close it right now :)

aqreed, 2016-11-08:
it is still using the old examples (Cessna 310)...

Juanlu001, 2016-11-08:
The current link shows a static version using nbviewer, still outdated. These are two separate issues: one involves changing one line, the other exporting all the scripts again. If someone does both, great! Otherwise I don't see a reason not to restore the link to mybinder, now that it's working.

Juanlu001, 2016-11-08:
Closed in #94!

closed 2016-11-08 by Juanlu001