AeroPython/PyFME #92: Plot Upgrades

salgozino, 2016-11-07:
I have some idea to upgrade in the plots that the simulations output.

- [] I think that is important to plot the evolution of the deflection angles of the commands to understand why are changing all the other variables.

- [] The units in the plot are importants. I guess that the angles are in radians, the forces and moments are in Newtons and N.m but it's just a suposition due to the order of magnitude of the values. The coordinates in Earth are in meters?

AlexS12, 2016-11-07:
Hi @salgozino, thanks for the issue.

You are right! in the near future, one of the main tasks should be implementing a module to deal with the outputs properly. As you have mentioned it would be interesting to plot some other variables and right now, there is no obvious and easy way to do it. 

Of course, units, titles and legends are fundamental.  Answering your question, forces are expressed in N and moments in N·m :wink:

aeroaks, 2016-11-14:
I would be interested working on this! though would need clarity on the desired result!

AlexS12, 2017-11-13:
Conversation will continue in #58

closed 2017-11-13 by AlexS12