AeroPython/PyFME #95: Update notebook examples

Juanlu001, 2016-11-08:
After merging #79, the example scripts and the example notebooks do not match anymore. We should keep only the notebook format, therefore:

* [ ] Converting the `.py` scripts in `examples` to Jupyter notebooks
* [ ] Remove the old `examples/examples-notebook` directory and the `.py` scripts in `examples`
* [ ] Make sure these work from mybinder by adding a command to install PyFME, see for instance https://github.com/poliastro/poliastro/blob/1c3c1da/index.ipynb

aqreed, 2016-11-14:
Ok I didn't check this before as I thought it did not include deleting the `.py`

@aeroaks great job anyway with those nb!

aeroaks, 2016-11-14:
@aqreed Aqr thanks!

AlexS12, 2017-11-13:
After the merge of #104 The example notebooks will disappear as we know them now.

In my opinion, adding more examples is a good idea, but instead of having many files with different maneuvers, we could focus on different topics that let the user understand the possibilities of the library:

* Notebook with standard workflow of a simulation
* Notebooks explaining internal working of main classes and their interfaces with others.
* Notebooks on maneuvering capabilities flight envelope using the trimmer
* Notebooks with different maneuvers, but adding some kind of comparison with theoretical simplified expressions from bibliography or empirical values

If you have any idea and some time to work on this, we can keep this issue opened and keep talking :muscle:

acelere, 2017-11-17:
I think that the notebook example is much better now. Thank you. 
Only one suggestion: if you could add a small cell where you show how to propagate for only one time step, I think this will help in debugging and also, in my case, when implementing this nice model with other flight control modules.

AlexS12, 2017-11-18:
Thanks for having a look @acelere! :tada: 
I think it is a great idea to add some cells explaining how to propagate for one time step. Then, you can check what's going on, and continue with another time step using propagate again (that will continue integration from the state last propagation generated).

I have just added an example at the end of the notebook! Does it meet your needs? :+1: