AeroPython/PyFME #96: Rudder aerodynamic coefficients *not convincing*

aqreed, 2016-11-08:
As it can be noted in [lines 277 and 284](https://github.com/AeroPython/PyFME/blob/master/src/pyfme/aircrafts/cessna_172.py#L277), the coefficients Cl_delta_rud and CN_delta_rud have been tuned (multiplied by 0.075). This was done to get a "decent" behaviour when deflecting the rudder, as a rough guess.

While this works, it is far from convincing, to the point I can barely sleep at night thinking on it.

### Origin
The formulae to calculate Cl_delta_rud and CN_delta_rud can be found from Roskam's [Methods for Estimating Stability and Control Derivatives of Conventional Subsonic Airplanes](https://app.box.com/s/7k03cfwc06gxuwtqkuwbpruxu67p2v0z) pages 12.1 and 12.2. Both coefficients heavily depend on CY_delta_rud and alpha.
### The question
Does anybody know how to improve this? What are we doing wrong?

AlexS12, 2016-11-08:
We have talked about this some times before and regarding the plane is only big issue I can see :wink: 

There are two possibilities (and maybe it is not possible, but the exercise could be fruitful and gratifying):

- As we could find somo information about the dumpings, frequencies and phases of this aircraft's modes. We could tune the parameters following an iterative process (we could think how to automatize it with some kind of optimization, GA @AunSiro ...):
    * Tune the parameters
    * Simulate exciting one mode
    * Obtain mode's parameters for the simulated values
    * Repeat until tuned and recovered match. 

- There is bibliography about calculating the aircraft's response to the inputs for short times after the input using the linearized models (MA. Gomez Tierno et al - Mecánica del Vuelo, I will try to find some references in English). Maybe we can find the transfer functions for this aircraft (it's common to use the in the design of control systems) and try to reproduce the values adjusting our "tunned" parameters.

Do you think it is feasible? :thinking:

salgozino, 2016-11-11:
For references in English for the aircraft's response I use:
Stengel - Flight Dynamics (This one I think is the best)
Roskam - Airplane Aerodynamics And Performance
Nelson - Flight Stability And Automatic Control (@aqreed Here you can find basic and quick aproximations for the aerodynamic coefficients too! ;-) )

I'm performing some calculation to understand why tou have to use this correction factor in the rudder coefficients. Let me finish these numbers and I will be writting here my comments

aqreed, 2016-11-11:
My guess is that the calculation of CY_delta_rud must be wrong! Regarding control theory and flight dynamics I would also add:
- Dynamics of Flight, B. Etkins (a classic!)
- Automatic Flight Controls Systems, D. McLean
- Fundamentals of Airplane Flight Mechanics, D. Hull
- Automatic Controls of Aircraft and Missiles, J.H. Blakelock

All of them are good, choose the one that you prefer!

salgozino, 2016-11-23:
@aqreed I was checking the CY_delta_rud_data and Cl_delta_rud_data of [https://github.com/AeroPython/PyFME/blob/master/src/pyfme/aircrafts/cessna_172.py#L277](url) and that values does not match with respect to the values that could be found in Roskam Airplaine Design Part VI Chapter 11 Page 493. 
In that reference CY_delta_rud = 0.187 and Cl_delta_rud = 0.0147.
In the aircraft file of the PyFME CY_delta_rud = 0.561 and Cl_delta_rud (for alpha = 0) is 0.0463.

Maybe, the estimation of the CY_delta_rud must be rechecked as you said. I will try to reproduce that estimations to compare with the values that are posted in the aircraft file.

AlexS12, 2016-11-23:
@salgozino Thank you @salgozino for your support with such a meticulous observation. Hopefully, will get better understanding of the problem and find a suitable approximation soon!

AlexS12, 2017-11-13:
Hi @salgozino, did you finally accomplish any estimation for these coefficients?