AeroPython/PyFME #102: Build a new aircraft model ?

liubenyuan, 2016-12-16:
How could I build a new aircraft model for PyFME? For example, a F-16 falcon as in BMS 4.2

aqreed, 2016-12-21:
Hello @liubenyuan, sorry for the delay.

You have to create an input file for DATCOM, you can start by modifying to [this one](https://github.com/AeroPython/PyFME/blob/master/data/cessna172.inp) for the C172. Actually it would be better if you follow [this](http://wpage.unina.it/agodemar/DSV-DQV/DATCOM_Tutorial_I.pdf) tutorial.

TsingQAQ, 2017-08-30:
Hi @aqreed and others,

   I'm new to this program, and looks like it has some promising features. I wonder if you have other ways to build an aircraft data when I have all the datas already, like I've got an aircraft's all aerodynamic,stability and control datas due to other solver(not Datcom, maybe some other CFDs). Is there a way todo so?

AlexS12, 2017-08-30:
Hi @TsingQAQ, 

Currently we implement the aircraft directly in a python class with aerodynamic data hard coded there. However, we are really looking forward to implementing loaders that can import the aircraft data from standard data formats. 

We are open to pull requests and any advice or suggestions in this respect (i.e. Advisable standards to be compatible with)

aeroaks, 2017-08-30:
@AlexS12 Would definitely like to help with these!!! Count me in

TsingQAQ, 2017-08-30:
Thanks for your fast reply @AlexS12,  
    Actually I'm looking to find some code to do a Flight Dynamics calculation. I'm a beginner of this area and still looking around for some code to do a job like this. Though I'm a python user, I'm still a little bit puzzled since flight dynamics calculation needs high performance calculation, the famous[ JSBsim](http://jsbsim.sourceforge.net/) are written in C++, so how's the performance of a python code which does the similar job?

AlexS12, 2017-08-30:
@TsingQAQ If you need reliable calculations I recommend you using jsbsim. PyFME is not yet validated and mature. Pyfme is expected to be slower than any c++ implementation, at least until he have a reliable code and can focus on performance (i.e. Numba jit)

TsingQAQ, 2017-08-31:
@AlexS12 Many thanks for your suggestion.
 I'll keep looking on pyFME and hopefully it will become a mature and nice tool!

AlexS12, 2017-11-13:
Hi, @TsingQAQ. Did you finally go any further with your aircraft model? did you try jsbsim finally? It would be interesting for us to know! :airplane:

gurkanctn, 2018-07-17:
@AlexS12 , "Currently we implement the aircraft directly in a python class with aerodynamic data hard coded there. However, we are really looking forward to implementing loaders that can import the aircraft data from standard data formats."

The script can read and load data from a "parameters" TXT file, which would be fairly human editable and machine readable.

In the beginning, the current hardcoded values can be read from the Parameters file.

After that, aircraft (parameters file) selection can be considered.

And overriding/manipulating files can be a further feature to optimize the parameters (e.g. to deliver certain flight characteristics.).