AeroPython/PyFME #103: YSFlight simulator support

Symbian9, 2017-01-03:
YSFlight - freeware minimalistic flight simulator with open file formats.

ABOUT
* http://ysflight.org
  * http://www.vskylabs.com/2015/10/jetmanhuss-ysflight-report-2015.html?m=1

File formats descriptions and tools
* https://www.sites.google.com/site/ysdecaff/ysflight-scientific-research
  * https://sites.google.com/site/ysdecaff/decaff-s-ysflight-tools
* https://forum.ysfhq.com/viewtopic.php?t=8172
* http://github.com/ysflight-opensource

AlexS12, 2017-11-13:
Hi @Symbian9, thanks for the issue and sorry for such a late response. It would be interesting for us to have a visual output taking advantage of an already developed flight simulator. 

I had a look at the links you provided and I have some questions:

* I was able to find the dat file definition for the aircrafts (https://forum.ysfhq.com/viewtopic.php?f=33&t=1772 and http://ysflightsim.wikia.com/wiki/DAT_Files) are the official aircraft dat files available somewhere?
* I have read that it has its own flight mechanics engine, is it open? I have not been able to find any code.
* Is there any specification and/or example on how an external software should communicate with YSFlight Simulator? For us it would be interesting to provide the aircraft state and be able to see the aircraft flying.

Symbian9, 2017-11-14:
@AlexS12, hi!

> are the official aircraft dat files available somewhere?

Official aircrafts packaged with YSFlight, so if you already has installed YSFlight then you already got official addons DAT's (users of YSFlight call it as "stock" addons/pack). You can get YSFlight here:
* http://ysflight.org/download

> I have read that it has its own flight mechanics engine, is it open? I have not been able to find any code.

YSFlight mechanics is not open, but mostly reverse engined by *Decaff* (aka *Decaff_42*, *YSDecaff*)
* https://sites.google.com/site/ysdecaff/ysflight-scientific-research

Also here are few Python-based tools for analyze flight record files (`.yfs` format) produced by YSFlight
* https://sites.google.com/site/ysdecaff/decaff-s-ysflight-tools

> Is there any specification and/or example on how an external software should communicate with YSFlight Simulator? For us it would be interesting to provide the aircraft state and be able to see the aircraft flying.

Yes, there are such info about YSFlight and all related to it tools:
* "YSFlight Handbook" (free PDF book provided by YSFHQ community, focused on YSFlight internals, format specifications and addon development), WIP
  * https://forum.ysfhq.com/viewtopic.php?f=147&t=8172#p92286
  [image]
* Open-source tools on Github
  * https://github.com/ysflight-opensource
* OpenYS (open-source server app for YSFlight in `C#`)
  * https://forum.ysfhq.com/viewtopic.php?t=7589
* YS ATC Radar (open-source Python3.4/Tkinter app), WIP
  * https://forum.ysfhq.com/viewtopic.php?t=8492
  [image]

I will be happy to help with any questions related to YSFlight! ;-)

AlexS12, 2017-11-14:
Thank you for such a quick response! We will have a deeper look at these references and take YSFlight into account among other options such as Flight Gear.

Integration with a flight simulator is not currently in our short term scope unless we incorporate someone with more experience on this kind of APIs to the project so I will keep this issue open just as a reminder of our long term intentions in case someone interested passes by here