AeroPython/PyFME #108: Trimming condition at 2g

juanmavaz, 2018-04-13:
How do you set up the environment to trim the airplane at 2 g's?

Thanks

AlexS12, 2018-04-16:
Hi! 
Currently, the only inputs you can fix for the trimming process are: 

```
    pos : Position
        Initial position of the aircraft.
    psi : float, opt
        Initial yaw angle (rad).
    TAS : float
        True Air Speed (m/s).
    controls : dict
        Initial value guess for each control or fixed value if control is
        included in exclude.
    gamma : float, optional
        Flight path angle (rad).
    turn_rate : float, optional
        Turn rate, d(psi)/dt (rad/s).
```

This allows you to trim the aircraft in a stationary condition. It is possible that in the trimmed condition the aircraft normal acceleration is 2g (ie. a turn with small turning radius), but currently it is not possible to choose the acceleration as input for the trimming process.

In my opinion that is more related with a control system or autopilot (closed loop). Anyway, do you have any idea about how the trimming process would work in that case? We could have a look at it!

The current trimmer code is this: https://github.com/AeroPython/PyFME/blob/master/src/pyfme/utils/trimmer.py
And the implementation is inspired in: Stevens, BL and Lewis, FL, "Aircraft Control and Simulation", Wiley-lnterscience.

juanmavaz, 2018-04-17:
Hi Alex,

I had my own simulation routines for aircraft loads purposes, that is why I'm interested in other trimmed conditions apart for the 1g. When I found PyFME, I liked it, and wanted to implement the aircraft model I already had to check how it works.

What I've done is to modify the VerticalConstant class by including an argument when creating a new object. The argument just multiplies the force, so you get the trimming condition you were looking for.

Thanks!

closed 2018-07-16 by AlexS12

gurkanctn, 2018-07-17:
This document has some explanations regarding trim. @AlexS12 

http://wpage.unina.it/agodemar/DSV-DQV/AIAA-2007-6703-905_DeMarco_Duke_Berndt.pdf