AeroPython/PyFME #118: bug

parthp08, 2019-06-10:
https://github.com/AeroPython/PyFME/blob/d1bbc0a178440cf1bd333c4837bd1bb07a3323ad/src/pyfme/utils/trimmer.py#L239

value of  r should be r = turn_rate * cos(theta) * cos(phi)  not r = turn_rate * cos(theta) * sin(phi) in trimmer.py

parthp08, 2019-06-10:
source : Aircraft Control and Simulation stevens book page no. 342

AlexS12, 2019-06-10:
Thanks for reporting, @parthp08. You are right! The equation is wrong. Would you like to submit a PR or do you prefer someone else to fix it?

By the way, I saw you are developing a nice julia repo about Flight Dynamics based mainly in Stevens (https://github.com/parthp08/Aircraft-Control-and-Simulation/tree/master/3DoF%20Transport%20Aircraft). I am working on a similar one (https://github.com/AlexS12/FlightMechanics.jl). Would you like to talk about joining efforts at some point or maybe you prefer to continue on your own?

parthp08, 2019-06-10:
I already saw your work on FLightMechanics.jl and its very good and i would very much like to join effort as there is no point in creating similar stuff separately.

parthp08, 2019-06-10:
submitted PR for it.

parthp08, 2019-06-10:
@AlexS12  you can contact me on parthviradiya08@gmail.com if you like.

AlexS12, 2019-06-10:
Opened an issue here just in case we find more interested/interesting people https://github.com/AlexS12/FlightMechanics.jl/issues/34

Let's continue there if you don't mind!

closed 2019-06-10 by AlexS12