AeroPython/PyFME #120: Wind implementation

bboynton97, 2020-04-07:
This library is exactly what I need for a project I am working on with a team. The only drawback is the lack of support for other wind types.

I know this library isn't being supported anymore, but I would love if someone from the AeroPython team would outline what work has been done and how we can best work together to fork the repo and create some wind classes, including static wind, gusts, thermals and turbulence.

AlexS12, 2020-04-11:
Hi @bboynton97! Supporting different wind types is something we had in mind, but never accomplished (https://github.com/AeroPython/PyFME/issues/99):

- There is a wind module, where new classes could be added: https://github.com/AeroPython/PyFME/blob/master/src/pyfme/environment/wind.py
- Wind is taken into account in the aircraft class (https://github.com/AeroPython/PyFME/blob/master/src/pyfme/aircrafts/aircraft.py#L111) so knowing inertial velocity and wind, aerodynamic velocity can be calculated.

As you noticed, the repo is unmantained now. Our  longterm plan was to create a platform where different people could experiment plugging new things and tunning the existing ones (ie. more complex ac models, different dynamic systems, wind models, plugging a ocntroller such as an autopilot or a stability augmentation system), We worked until the point of having a complete, simple simulator, but didn't go ahead on extending the features.

Although the projects has some tests, IMO, some other validations should be done. For example, #113 remained open (my fault and I am sorry about that. It was a valuable and useful work). Of course, including new models such as the wind types you mention is a great feature.

You can have a look at the open issues if you need more ideas. Notice that there is also a [wiki](https://github.com/AeroPython/PyFME/wiki) (unfortunately, some pages are in Spanish, but I think that some interesting info is in English.

I encourage you to make a Fork of the repo and continue the effort if you are interested. Just for my curiosity, what kind of project do you have in mind? It's always great to know how people are using PyFME

bboynton97, 2020-05-01:
We built an OpenAI gym for unsupervised machine learning on PyFME. It's host on GitLab [here](https://gitlab.com/zbalda/gym-pyfme/)

AlexS12, 2020-05-01:
Wow! Seems amazing! It’s great to see such a great project using PyFME