AeroPython/PyFME #126: How to change the controls during one simulation

YingxiaoKong, 2020-05-13:
Hi I'm new to this package. I just installed it and it works pretty well. However, I'm wondering whether there is a way to change the controls during the simulation?  One way I can think about it to let the simulator to first simulates like 10 secs, Then you change the controls, then let it to simulate for another  10 secs. But the sim.propagate only returns the results, if I want to continue the simulation the previous 10 secs, I need to know the current state. Is there a way that I can get the current state? I checked the results file, it contains: 
'temperature', 'pressure', 'rho', 'a', 'Fx', 'Fy', 'Fz', 'Mx', 'My',
       'Mz', 'TAS', 'Mach', 'q_inf', 'alpha', 'beta', 'rudder', 'aileron',
       'elevator', 'thrust', 'x_earth', 'y_earth', 'z_earth', 'height', 'psi',
       'theta', 'phi', 'u', 'v', 'w', 'v_north', 'v_east', 'v_down', 'p', 'q',
       'r'

But the full-state should also contains : u_dot: 0.00 m/s², v_dot: -0.00 m/s², w_dot: 0.00 m/s² 
P_dot: 0.00 rad/s², Q_dot: -0.00 rad/s², R_dot: -0.00 rad/s² 

Is there a way that I can find the above parameters?

Besides, I'm still not very sure of that theta and phi are. It's said they're eular angles. Is theta the pitch angle? Is there any way I can set the pitch angle? Or in this aircraft model, the pitch angle doesn't matter a lot.

AlexS12, 2020-05-15:
Hello, @YingxiaoKong I will try to answer your questions one by one. Hope it helps!!

> Hi I'm new to this package. I just installed it and it works pretty well. However, I'm wondering whether there is a way to change the controls during the simulation? One way I can think about it to [...]

I understand. We never got to implement the control loop. The only feasible way I can imagine to do that now is to change the simulation controls attribute after the first 10s (`sim.controls = ...`)

> But the sim.propagate only returns the results, if I want to continue the simulation the previous 10 secs, I need to know the current state. Is there a way that I can get the current state? I [...]

There is an argument to control what is saved during a simulation. https://github.com/AeroPython/PyFME/blob/master/src/pyfme/simulator.py#L42 

I think you could get what you need from `system.full_state.acceleration` and `system.full_state.angular_accel` (https://github.com/AeroPython/PyFME/blob/master/src/pyfme/models/state/aircraft_state.py#L34)

> Besides, I'm still not very sure of that theta and phi are. It's said they're eular angles. Is theta the pitch angle? Is there any way I can set the pitch angle? Or in this aircraft model, the pitch angle doesn't matter a lot.

We follow Tait-Bryan convention (ZYX rotation order) for attitude
theta -> pitch (y rotation)
phi -> roll (x rotation)

To change the pitch angle during the simulation you will need to modify the controls. Elevator will be the one affecting the most

YingxiaoKong, 2020-05-15:
Hi,
Thank you so much for your reply!!

I’m going to try your suggestions and I will let you know if there is further questions.

By the way, the installation suggestion by using’ pip install...’ seems has some problem. I tried that command first and the package I downloaded was different from the one in github. So I tried it again using ‘setup.py’ and the one is exactly the same as github version. That’s a tiny issue but I just want to let you know in case other people may have encountered the same issue.

YingxiaoKong, 2020-05-19:
I tried the following code and it works!

As you suggested, I modified' _default_save_vars' by adding:
        'u_dot':'system.full_state.acceleration.u_dot',
        'v_dot':'system.full_state.acceleration.v_dot',
        'w_dot':'system.full_state.acceleration.w_dot',
        'p_dot':'system.full_state.angular_accel.p_dot',
        'q_dot':'system.full_state.angular_accel.q_dot',
        'r_dot':'system.full_state.angular_accel.r_dot',

Then I simulated 10 seconds, and got the results named 'results'. After this  I obtained the last state by the following code:

def obtain_last_state(results):
    from pyfme.models.state import AircraftState, EulerAttitude, BodyVelocity
    from pyfme.models.state.acceleration import BodyAcceleration
    from pyfme.models.state.angular_velocity import BodyAngularVelocity
    from pyfme.models.state.angular_acceleration import BodyAngularAcceleration

    last_val=results.iloc[-1]
    last_theta = last_val['theta']
    last_phi = last_val['phi']
    last_psi = last_val['psi']
    new_att = EulerAttitude(theta=last_theta, phi=last_phi, psi=last_psi)

    last_u = last_val['u']
    last_v = last_val['v']
    last_w = last_val['w']
    new_vel = BodyVelocity(u=last_u, v=last_v, w=last_w, attitude=new_att)

    last_x = last_val['x_earth']
    last_y = last_val['y_earth']
    last_height = last_val['height']
    new_pos = EarthPosition(x=last_x, y=last_y, height=last_height)

    last_p = last_val['p']
    last_q = last_val['q']
    last_r = last_val['r']
    new_angular_vel = BodyAngularVelocity(last_p,last_q,last_r,new_att)

    last_u_dot = last_val['u_dot']
    last_v_dot = last_val['v_dot']
    last_w_dot = last_val['w_dot']
    new_acceleration = BodyAcceleration(last_u_dot,last_v_dot,last_w_dot,new_att)

    last_p_dot = last_val['p_dot']
    last_q_dot = last_val['q_dot']
    last_r_dot = last_val['r_dot']
    new_angular_accel = BodyAngularAcceleration(last_p_dot,last_q_dot,last_r_dot,new_att)
    new_state = AircraftState(new_pos,new_att,new_vel,new_angular_vel,
                         new_acceleration,new_angular_accel)
    return new_state
new_state = obtain_last_state(results)

I'm not sure whether this is the quick way to obtain the state of the last seconds, though. 
I simulated for another 10 seconds and obtained the results. I compared the last second's result of this one and the one by directly simulating for 20 seconds and they're the same! So it works!

closed 2020-05-20 by YingxiaoKong